RosettaDrone/rosettadrone #113: Waypoint Action like Camera TAKE_PHOTO not working

Phillweston, 2022-06-08:
**Describe the bug**
When using waypoint actions like taking photos in waypoint, an error occurs after uploading the mission plan:

**To Reproduce**
Steps to reproduce the behavior:
1. Click on 'Fly'
2. Add 'Takeoff' point
3. Add 'Waypoint' with camera action like Take photo
4. See error

**Expected behavior**
Waypoint uploaded and ready to execute.

**Screenshots**
APP
[image]
QGC
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - QGC: Version 4.2.1

**Smartphone (please complete the following information):**
 - Device: DJI Mavic 2 Pro with smart controller
 - DJI SDK Version: DJI SDK 4.16 

**Additional context**
Please help me with this problem, thanks!

Phillweston, 2022-06-08:
In my mission above, the total distance is only 76m, far less than 500m. I think this is a bug related to the waypoint mission handling.

kripper, 2023-02-17:
You can also try the new MissionManager (must be enabled in the code, search for "useMissionManager")

kripper, 2023-02-21:
Let me know if you need help.

closed 2023-02-21 by kripper