RosettaDrone/rosettadrone #120: MAVSDK and Rosetta compatibility

kripper, 2023-02-20:
The goal of this issue is to get `fly_qgc_mission.cpp` from MAVSDK to start a mission on Rosetta.

Before that, I managed to get `fly_qgc_mission.cpp` to start a mission on Arducopter SITL using some workarounds described here: https://github.com/ArduPilot/ardupilot/issues/22966.

I have been using Arducopter SITL as a reference for testing and making Rosetta compatible with QGC, which is now done, ie. we can send missions from QGC to Rosetta.

But now, when I tro to send a mission from `fly_qgc_mission.cpp` to Rosetta, I don't even see the MAVlink commands reaching Rosetta, as if `fly_qgc_mission.cpp` is only making a connection, but not sending anything.

This is the output I get on the MAVSDK side:

```
Waiting to discover system...
Discovered autopilot
sending again after 0.50748 s, retries to do: 3  (520).
sending again after 0.508117 s, retries to do: 3  (511).
sending again after 1.01818 s, retries to do: 2  (520).
sending again after 1.01875 s, retries to do: 2  (511).
sending again after 1.51954 s, retries to do: 1  (520).
sending again after 1.52036 s, retries to do: 1  (511).
Retrying failed (520)
Retrying failed (511)
sending again after 10.0052 s, retries to do: 3  (400).
sending again after 10.0057 s, retries to do: 3  (512).
Request was for msg ID: 242
sending again, retries to do: 3  (INS_GYROFFS_X).
sending again, retries to do: 2  (INS_GYROFFS_X).
sending again after 20.0158 s, retries to do: 2  (512).
Request was for msg ID: 242
sending again after 20.0164 s, retries to do: 2  (400).
sending again after 30.0209 s, retries to do: 1  (400).
sending again after 30.0216 s, retries to do: 1  (512).
Request was for msg ID: 242
sending again, retries to do: 1  (INS_GYROFFS_X).
Error: Retrying failed get param busy timeout: INS_GYROFFS_X
Error: Param for gyro offset_x failed.
Retrying failed (512)
Retrying failed (400)
Arming failed: Timeout
```

And this is the basic code I'm running inside `fly_qgc_mission.cpp`:

```
    ...
    const ConnectionResult connection_result = mavsdk.add_any_connection(argv[1]);
    if (connection_result != ConnectionResult::Success) {
        std::cerr << "Connection failed: " << connection_result << '\n';
        return 1;
    }

    auto system = get_system(mavsdk);
    if (!system) {
        return 1;
    }
    auto action = Action{system};
    auto mission_raw = MissionRaw{system};
    auto telemetry = Telemetry{system};

    if(1) {
        // TESTS:

        // Increase timeout (just in case)
        mavsdk.set_timeout_s(10);

        // Enable logging
        mavsdk::log::subscribe([](mavsdk::log::Level level,   // message severity level
                                const std::string& message, // message text
                                const std::string& file,    // source file from which the message was sent
                                int line) {                 // line number in the source file
            std::cout << message << "\n";

            // returning true from the callback disables printing the message to stdout
            return level < mavsdk::log::Level::Warn;
        });

        // Try to arm
        const Action::Result arm_result = action.arm();
        if (arm_result != Action::Result::Success) {
            std::cerr << "Arming failed: " << arm_result << '\n';
            return 1;
        }
        std::cout << "Armed.\n";

        return 0;
    }

```

I'm running the MAVSDK code using `udp://:14551`

I also added a dummy class to Rosetta which makes it easy to start the app on Android without needing to connect to a drone.
Just press 5 times on the logo on the first screen and you will be able to start in TestMode.
The MAVLink IP and port can be configured in the settings (top right).

The dummy class will send hardcoded telemetry which is just enough to test.

BTW, the result is the same when connected to a real drone.
And as I said before, Rosetta works fine QGC.

kripper, 2023-02-21:
Same results running mavsdk_server on another machine:

```
mavsdk-windows-x64-release\bin\mavsdk_server_bin.exe udp://:14550
[09:10:19|Info ] MAVSDK version: v1.4.12 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:20)
[09:10:19|Info ] Waiting to discover system on udp://:14550... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
[09:10:20|Info ] New system on: 192.168.0.5:34724 (with sysid: 1) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\udp_connection.cpp:192)
[09:10:20|Debug] New: System ID: 1 Comp ID: 1 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:496)
[09:10:20|Debug] Component Autopilot (1) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:377)
[09:10:20|Warn ] Vehicle type changed (new type: 2, old type: 0) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:225)
[09:10:20|Debug] Discovered 1 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:578)
[09:10:20|Info ] System discovered (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:63)
[09:10:20|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:53)
[09:10:20|Info ] Server set to listen on 0.0.0.0:50051 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:54)
[09:10:20|Warn ] sending again after 0.506164 s, retries to do: 3  (520). (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavlink_command_sender.cpp:287)
[09:10:21|Warn ] sending again after 1.01653 s, retries to do: 2  (520). (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavlink_command_sender.cpp:287)
[09:10:21|Warn ] sending again after 1.52467 s, retries to do: 1  (520). (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavlink_command_sender.cpp:287)
[09:10:22|Error] Retrying failed (520) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavlink_command_sender.cpp:307)
```

kripper, 2023-02-21:
I found the problem. Rosetta was not supporting MAVLink 2.
I'm implementing it right now.

closed 2023-02-21 by kripper

julianoes, 2023-02-21:
Aha!

kripper, 2023-02-21:
Fixed.