RosettaDrone/rosettadrone #122: Transmir the raw DJI video stream to an external decoder

kripper, 2023-02-24:
This post is to discuss if it makes sense to send the raw data received by the DJI SDK to an external ffmpeg node and do the decoding outside Rosetta.

Why?

- The external decoder could just decode frames when needed. For example, an AI script could need less frames but with higher quiality.
- It would make it easier to try and fine-tune different decoding alternatives: software, GPU hardware and drivers, ffmpeg options, etc.
- The decoding could be done on external more powerful hardware.

Problems:

I implemented a simple UDP transmiter of the raw video data received from the SDK.
On the end point, I failed to decode the video with ffmpeg altough fmpeg is used by Rosetta to decode it and stream it via RTP.

At least, `ffprobe` seems to detect it. This is the output for a DJI Mini SE:

```
[STREAM]
index=0
codec_name=h264
codec_long_name=H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
profile=High
codec_type=video
codec_tag_string=[0][0][0][0]
codec_tag=0x0000
width=1280
height=720
coded_width=1280
coded_height=720
closed_captions=0
film_grain=0
has_b_frames=1
sample_aspect_ratio=1:1
display_aspect_ratio=16:9
pix_fmt=yuv420p
level=40
color_range=tv
color_space=bt709
color_transfer=bt709
color_primaries=bt709
chroma_location=left
field_order=progressive
refs=1
is_avc=false
nal_length_size=0
id=N/A
r_frame_rate=30000/1001
avg_frame_rate=89/3
time_base=1/1200000
start_pts=N/A
start_time=N/A
duration_ts=N/A
duration=N/A
bit_rate=N/A
max_bit_rate=N/A
bits_per_raw_sample=8
nb_frames=N/A
nb_read_frames=N/A
nb_read_packets=N/A
extradata_size=58
DISPOSITION:default=0
DISPOSITION:dub=0
DISPOSITION:original=0
DISPOSITION:comment=0
DISPOSITION:lyrics=0
DISPOSITION:karaoke=0
DISPOSITION:forced=0
DISPOSITION:hearing_impaired=0
DISPOSITION:visual_impaired=0
DISPOSITION:clean_effects=0
DISPOSITION:attached_pic=0
DISPOSITION:timed_thumbnails=0
DISPOSITION:captions=0
DISPOSITION:descriptions=0
DISPOSITION:metadata=0
DISPOSITION:dependent=0
DISPOSITION:still_image=0
[/STREAM]
[FORMAT]
filename=udp://192.168.0.5:5600
nb_streams=1
nb_programs=0
format_name=h264
format_long_name=raw H.264 video
start_time=N/A
duration=N/A
size=N/A
bit_rate=N/A
probe_score=51
[/FORMAT]

```

Please comment.

The1only, 2023-02-26:
DJI sometimes uses very large data blocks/streams that are hard to transmit over UDP. So we needed to reframe. If not the receiver will receive frames with no header as the network is allowed to split up UDP packages into several smaller ones. Even UDP superblocks are too small. So I found no solution.

kripper, 2023-02-26:
I see. I took the DJI video sample app (latest version) and added Rosetta's NAL splitter + RtpSocket sender sending the frames modified by the sample, and got the same result (video is received with heavy errors).

The sample adds a keyframe that is obtained differently using the SDK and adds it before the other frames obtained by the JNI code.

Please note that Rosetta never receives a keyframe (at least not for the Mini SE).

To obtain this keyframe, the sample uses a "magic" number of a resource ID that varies on each model. It's very strange what it does but it works on the screen. I can publish this forked sample and my commit with minimal modifications if someone is interested since this could be a good test case to isolate and test all our video streaming issues.

Then I tried to reduce the amount of data sent via RTP as suggested [here](https://github.com/roguestarslade/DJI-Android-VideoStreamDecodingSample/issues/1) but it didn't help either.

I believe there may be a problem between the NAL splitter code and gstreamer for the video format managed by the Mini.

BTW, where did you get the NAL splitting code from?

kripper, 2023-02-26:
Here is the updated DJI video sample app with Rosetta's RTP streamer:
https://github.com/kripper/DJI-Android-VideoStreamDecodingSample

This is the minimal commit to add Rosetta's RTP streamer to the sample:
https://github.com/kripper/DJI-Android-VideoStreamDecodingSample/commit/9d31891398c412b589f5820fc52cebc519c2ac5c

kripper, 2023-02-26:
I updated the code to enable streaming RTP with Unicast (Multicast didn't work for us).
https://github.com/roguestarslade/DJI-Android-VideoStreamDecodingSample/commit/6924305079400747a7a16626fe717e6caf486b5a

In this commit, I added a boolean "splitNALs" that can be modified using the debugger to disable the NAL splitting during runtime.
I discovered that disabling this boolean after the client started to receive the stream, improved the quality significantly, which means that **we are probably doing something wrong in the NAL splitting code**.

Without NAL splitting, gstreamer was not able to open the stream at all.

kripper, 2023-04-07:
Closing. This approach didn't work. I'm not sure if the problem was caused by the network layer splitting UDP packets (because the package size was relatively small). I also tried sending via TCP, but then I wasn't sure how to receive and decode the data on the receiving end (ffmpeg or Gstreamer).

Anyway, all video code in Rosetta should be replaced with latest DJI's implementation, so please continue reading here:
https://github.com/RosettaDrone/rosettadrone/issues/131

kripper, 2023-08-15:
Other option would be to implement https://github.com/RosettaDrone/rosettadrone/issues/175

closed 2024-04-10 by kripper