RosettaDrone/rosettadrone #123: Video: Rosetta only streams RTP but not RTCP?

kripper, 2023-02-24:
In the initial commit of `RtpSocket.java` on this repo, all code regarding sending RTCP datagrams is commented out:

`//mReport.setDestination(dest, rtcpPort);`

Why is Rosetta not sending RTCP? Should we?

The1only, 2023-02-24:
I belive there was some issues and we were under time preasure so focus was to get something working. Also the entire MPEG transcoder was removed as I remember it. With todays phone that is more powerfull it might be a more flexible solution.Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-02-24:
Ok, I enabled handling RTCP (on the RTP port).

1) What is the use case for sending video to two ports **on the same** IP?
It makes more sense to send it to two different IPs.

2) When I extract a single frame from the RTP stream, the image is always fucked up at the same vertical line (if not the whole image). This is strange. Have you seen this before? It's a MINI SE.

closed 2023-02-24 by kripper

kripper, 2023-02-24:
In addition to my previous question on the use-case for two dest ports on the same IP, what happened with the `TRANSPORT_MULTICAST` option that was added to `RtpSocket.java`?

The1only, 2023-02-24:
1) The idea was to be able to send video to two processes without the delay of a mux. One was the AI process the other the GCS… was never used, did it caus I could 😵‍💫 made it easy to test AI while still having video on the GCS. You can also have two Mavlink streams. But we ended up using mavproxy.2) My problem was that DJI makes a video stream without having to consider block size or framing. This could not be sent over UDP , so we needed to reframe it. If I remember light we do not decode the video, only repack? One other problem with the DJI frames is that I frames are not sent as expected. I spent quite som etime on this. Also different DJI drones got different format, but you know that. I used Mavic 2 Pro, ant tested on Mavic Air and Inspire (very different).But you are right, at times we hadd issues, ended up with gray screen (on gstreamer) with som inverted movement, made the AI go crazy. But with perfect radio connection and GPU on a powerfull Qgroundcontrol PC we was able to perform all tests we had to. One involving several miles AI flying. We hade more problems with Air and Inspire, but never investigated.Best regardTerje Nilsen9Tek ASNorway

The1only, 2023-02-24:
Maybe I should revisit this project, I’d like to test all the fixes you have done.Hmm you are right… If I remember right we hade issues with multicast in our w-lan router? (Pure technical issue) And there was some thing about not being able to start/stop the stream? Hmm… I came from another American project that had multicast issues and decided to move away from it. Do remember back then there was no interest so I was prety mush all by my selfe (with a non drone software project team). And we hade a set of requirements that was quite spesific. We did fly the Mavic with AI remotly from another location in the world watching a big screen via cellphone and giving messages to the system. Had a loot of funn! What you make now is way more general and the way it shuld be. 👍Best regardTerje Nilsen9Tek ASNorway