RosettaDrone/rosettadrone #124: Seems I need Rosettadrone for a science project again.

The1only, 2023-03-10:
What is the general status these days?
I will use the Mavic 2 Pro with a smartcontroller, Dronekit, and python API, and a NVIDIA Orin GPU. 
I plan to compile the system from a MAC M1.
Any heads up?

The1only, 2023-03-10:
Perfect, the code compiled right out of the box. 

Did you ever find out about the Grade issue?  ( We must use an old Grade to be able to get video?)

The1only, 2023-03-10:
So if you in AndroidStudio accept all suggested updates, the app still works now?

The1only, 2023-03-10:
I added all updates and now can't compile. Ever seen this:

Can't determine type for tag '<macro name="m3_comp_assist_chip_container_shape">?attr/shapeAppearanceCornerSmall</macro>'

kripper, 2023-03-10:
Gradle updates? No, I haven't tested updating Gradle and have never seen this error.

The1only, 2023-03-10:
OK, not important, just tested.

The1only, 2023-03-10:
PS: the DJI SDK can be updated from Android 4.16.1 to Android 4.16.4
It requires one small fix only in Rosettadrone, the fix from DJI is: 

The US RID function does not work properly on the DJI RC Pro remote controller and some mobile phones, resulting in the aircraft unable to take off: Fixed.

The1only, 2023-03-10:
Something is strange, are you sure the master is working? 
I get a lot of Mavlink errors, and the app can not run, even if I compile fresh after git clone ???

The1only, 2023-03-10:
Did anyone use a DJI smart-controller lately ??? It can no longer run Rosettadrone somehow. 
At least on my built.

kripper, 2023-03-11:
Sorry, I don't have a smart controller to test.
I remember somebody posted an issue regarding the smart controller time ago.
Whatever problem you are experiencing, it's probably easy to fix if it worked once and broke somehow.

kripper, 2023-03-11:
https://github.com/The1only/rosettadrone/issues/105

kripper, 2023-03-11:
What errors are you seeing?

The1only, 2023-03-11:
I got it working, it's a balancing act on the dependencies. I spent some hours trying to find what needed updating and to what level. I did changes to app/build.gradle, a small change in dummyProduct.java ( to support DJI std 14.16.4 ), and a tiny fix in Manifest (not mandatory but more best practice) :-)
Seems it all works now, let me know if you want me to create a pull request. You are the boss now !

kripper, 2023-03-11:
Nice.
Yes, please. Let's keep our code bases up to date.

The1only, 2023-03-11:
I opened a "Discussions" page, better too small talk there...

closed 2023-04-03 by kripper