RosettaDrone/rosettadrone #129: App crash when click open and in test mode

Sasysan, 2023-03-16:
Hi, I am new to Andorid apps anyway i am able to build the app but when i try to run it crash all the time both when I plug in the dji mini 2 and when I try test mode. If someone can help me i would appreciate it. I get the following error:

> 'E/UDT-JNI: stopped udp
> E/getNetworkObject: network type:1
> D/AndroidRuntime: Shutting down VM
> E/AndroidRuntime: FATAL EXCEPTION: main
>     Process: sq.rogue.rosettadrone, PID: 14145
>     java.lang.IncompatibleClassChangeError: The method 'java.lang.Class java.lang.Object.getClass()' was expected to be of type interface but instead was found to be of type virtual (declaration of 'com.google.android.gms.dynamic.ObjectWrapper' appears in /data/app/sq.rogue.rosettadrone-6xqXF7rZNy05mYRqqHkyZQ==/base.apk)
>         at com.google.android.gms.dynamic.ObjectWrapper.unwrap(com.google.android.gms:play-services-basement@@18.0.2:4)
>         at com.google.android.gms.maps.zzau.onCreateView(com.google.android.gms:play-services-maps@@18.0.0:12)
>         at com.google.android.gms.dynamic.zad.zab(com.google.android.gms:play-services-base@@18.0.1:2)
>         at com.google.android.gms.dynamic.DeferredLifecycleHelper.zaf(com.google.android.gms:play-services-base@@18.0.1:1)
>         at com.google.android.gms.dynamic.DeferredLifecycleHelper.onCreateView(com.google.android.gms:play-services-base@@18.0.1:2)
>         at com.google.android.gms.maps.SupportMapFragment.onCreateView(com.google.android.gms:play-services-maps@@18.0.0:1)
>         at androidx.fragment.app.Fragment.performCreateView(Fragment.java:2963)
>         at androidx.fragment.app.FragmentStateManager.ensureInflatedView(FragmentStateManager.java:386)
>         at androidx.fragment.app.FragmentStateManager.moveToExpectedState(FragmentStateManager.java:281)
>         at androidx.fragment.app.FragmentLayoutInflaterFactory.onCreateView(FragmentLayoutInflaterFactory.java:140)
>         at androidx.fragment.app.FragmentController.onCreateView(FragmentController.java:135)
>         at androidx.fragment.app.FragmentActivity.dispatchFragmentsOnCreateView(FragmentActivity.java:319)
>         at androidx.fragment.app.FragmentActivity.onCreateView(FragmentActivity.java:298)
>         at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:791)
>         at android.view.LayoutInflater.createViewFromTag(LayoutInflater.java:741)
>         at android.view.LayoutInflater.rInflate(LayoutInflater.java:874)
>         at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
>         at android.view.LayoutInflater.rInflate(LayoutInflater.java:877)
>         at android.view.LayoutInflater.rInflateChildren(LayoutInflater.java:835)
>         at android.view.LayoutInflater.inflate(LayoutInflater.java:515)
>         at android.view.LayoutInflater.inflate(LayoutInflater.java:423)
>         at android.view.LayoutInflater.inflate(LayoutInflater.java:374)
>         at androidx.appcompat.app.AppCompatDelegateImpl.setContentView(AppCompatDelegateImpl.java:699)
>         at androidx.appcompat.app.AppCompatActivity.setContentView(AppCompatActivity.java:195)
>         at sq.rogue.rosettadrone.MainActivity.onCreate(MainActivity.java:713)
>         at android.app.Activity.performCreate(Activity.java:7383)
>         at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1218)
>         at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3256)
>         at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3411)
>         at android.app.ActivityThread.-wrap12(Unknown Source:0)
>         at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1994)
>         at android.os.Handler.dispatchMessage(Handler.java:108)
>         at android.os.Looper.loop(Looper.java:166)
>         at android.app.ActivityThread.main(ActivityThread.java:7529)
>         at java.lang.reflect.Method.invoke(Native Method)
>         at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:245)
>         at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:921)
> E/DJIUSBWifiSwitchManager: ****resetSwitchFromWifiFlag
> E/DJIUsbAccessoryReceiver: receiver destroySession
> I/Process: Sending signal. PID: 14145 SIG: 9'

kripper, 2023-03-16:
Seems to be related with Google maps.
Did you configure valid keys following the README?

kripper, 2023-03-16:
I don't think it's the same problem.
Did you maybe update gradle? Don't do it.

You could also try Terje's branch:
https://github.com/The1only/rosettadrone/tree/work_branch

He just updated versions.

kripper, 2023-03-16:
Maybe this:
https://issuetracker.google.com/issues/232131782?pli=1

closed 2023-03-16 by Sasysan

kripper, 2023-03-16:
@The1only did you also fix this on your working branch?
Seems that this was the problem.

I didn't merge yet, because I didn't notice the problem.
I have been busy with the precision landing implementation. Trying to bring together two top projects for the long term:
https://github.com/chobitsfan/apriltag_plnd/issues/1

The1only, 2023-03-16:
Yes I hade issues with Play services, took me a while to figure out to what level I could uppdate while stil be able to make things work.We support Android 5.1 and 7.1 that might be affected by this.TerjeBest regardTerje Nilsen9Tek ASNorway

chobitsfan/apriltag_plnd#1