RosettaDrone/rosettadrone #130: Questions about Rosettadrone wrapper working with M300 and OSDK

Shivam-Kataria, 2023-03-16:
Hi, I am new to rosettadrone, and I am working on a project where with DJI Matrice 300 where I would like to be able to control it using dronekit.

The project involves using the camera feed from the drone to decide where the drone will head to. To resolve the latency issues of having to stream video back to the ground control station, I have an onboard computer mounted on the drone for computation to occur. DJI has the Onboard SDK (OSDK) which is in c++ to control the aircraft.

My questions are:
1) Would this be possible to do that with M300 using rosettadrone?
2) Would it be possible to do that using an onboard computer instead of a ground control station?
3) Would it involve using the c++ OSDK or only dronekit?
4) How would I go on about achieving it with rosettadrone (I am a little bit of a newbie with this)?

The1only, 2023-03-16:
1. Yes I used Rosettadrone to control the Matrice 220 during my tests, the 300 is similar. 
2. Yes, however, it is quite some work, but there is a GitHub project that transforms DJI Onboard SDK to Mavlink. I believe it might be abandoned.
	I got one of the DJI onboard computers using Tegra 2 GPU, but never had time to complete the onboard SDK test…
3. You can do either but NOT both. 
4. With Rosettadrone you do NOT need to do anything on the drone itself. If you use Crystal Sky, then install Rosettadrone on it, and start-up. It will connect and you will be ready to set up the other side, 
	Connect the ground computer (I like to use a Linux-based computer of sort), then Start by getting QGC ( Ground Control) to work. Once you got that working you know everything is OK and can start your software.
	A hint is to use the Mavproxy to split the telemetry stream to several subscribers. That way you can have QGC and you’re AI software running at the same time.

Shivam-Kataria, 2023-03-16:
Thanks for the fast reply!

In the past, a past colleague of mine did try to use it without an onboard computer and the video lags were quite a bit so we have now decided to go with the onboard computer for less lags and faster computation.

The DJI Onboard SDK to Mavlink you mentioned, is that https://github.com/The1only/DJI-Onboard-SDK-for-Mavlink ? Looking at the repo, I am quite unsure on how it integrates together with python dronekit.

kripper, 2023-03-16:
> A hint is to use the Mavproxy to split the telemetry stream to several subscribers

Not necessary anyone.
I implemented multiple telemetry socket listeners.
You can now send mavlink commands from QGC and other (infinite) remote mavlink nodes.

I didn't commit yet, because I didn't test the clean disconnection code (threads). But I can do it now if you want to use and test it.

kripper, 2023-03-16:
@Shivam-Kataria for video related issues, please see https://github.com/The1only/rosettadrone/issues/131

@The1only can you please also test with you models and provide the exact gstreamer command line we should use to test the RTP streaming reception?

I cannot test, because my Mini SE seems to be the only model that doesn't work.

The1only, 2023-03-16:
I hade to custom compile gstreamer to be able to use GPU etc. when I did this the latency was no longer than the onboard SDK. Also remember to run without buffering that will create latency (obviously). Look at the video sample on the Rosettadrone page, it shows the full latency I had.Best regardTerje Nilsen9Tek ASNorway

The1only, 2023-03-16:
This is my video capture string from AI Python:

self.cap = cv2.VideoCapture('udpsrc port=5600 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264" ! rtph264depay ! avdec_h264 ! videoconvert ! appsink', cv2.CAP_GSTREAMER)

These are my test setting:

#Sende OK
/Library/Frameworks/GStreamer.framework/Commands/gst-launch-1.0 autovideosrc ! video/x-raw,framerate=20/1 ! videoscale ! videoconvert ! x264enc tune=zerolatency bitrate=3000 speed-preset=superfast ! rtph264pay ! udpsink host=127.0.0.1 port=5600

#Motta OK both with Rosetta and abou line...
/Library/Frameworks/GStreamer.framework/Commands/gst-launch-1.0 -v udpsrc port=5600 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! rtph264depay ! decodebin ! videoconvert ! autovideosink

kripper, 2023-03-16:
Ok, I added detailed instructions for testing here:
https://github.com/kripper/DJI-Android-VideoStreamDecodingSample/blob/develop/README.md

closed 2023-04-07 by kripper