RosettaDrone/rosettadrone #131: Replace all video related code with upstream code

kripper, 2023-03-16:
All video code in Rosetta should be replaced in the future with DJI's upstream maintained video code.
The code in Rosetta is outdated, difficult to maintain and is missing changes added by DJI.

Please test this fork with minimal Rosetta's changes for streaming RTP:
https://github.com/kripper/DJI-Android-VideoStreamDecodingSample

And please leave here a comment with the results of testing this fork with your specific drone model.

kripper, 2023-03-31:
@pktiuk is video working for you? If so, maybe you could try replacing our current 560 MB lib folder with these new files from upstream:
https://github.com/RosettaDrone/DJI-Android-VideoStreamDecodingSample/tree/develop/android-videostreamdecodingsample/libs

This way we could fix:
https://github.com/RosettaDrone/rosettadrone/issues/141

Video streaming never worked on my model (mini se), so I cannot test myself.

Replacing all old video code with upstream (this issue) would be ideal, but would probably requiere additional effort.

pktiuk, 2023-04-03:
> If so, maybe you could try replacing our current 560 MB lib folder with these new files from upstream

`lib` folder with removed unnecessary files works well.

I will try to test video soon.

pktiuk, 2023-04-03:
I am not able to test it with drone because of: #145 :(

pktiuk, 2023-04-06:
The first test was without attached camera.    
In test App after turning on the app (and pressing Open) I get screenshot from onboard camera. 
Later I was unable to get any other footage from this drone.
[testing_logs_logcat.log](https://github.com/RosettaDrone/rosettadrone/files/11167784/testing_logs.log)

The second with attached camera:   
 
`Demo TextureView` worked well, `Demo custom decoder` also worked but with a bit bigger latency, `Demo SurfaceView` also worked well   
`YUV Screen Shot` just showed gray rectangle. (I was seeing them on the controller screen)
[logcat_attached.log](https://github.com/RosettaDrone/rosettadrone/files/11167888/logcat_attached.log)
I was also able so get stream on my PC.

But later after some relaunches app broke (it was not showing any image)

[logcat_broken.log](https://github.com/RosettaDrone/rosettadrone/files/11167947/logcat_broken.log)

kripper, 2023-04-06:
My suggestions:

- First, try to figure out why the demo app works sometimes and sometimes not. It's only 0's and 1's, so there should be consistent behavior. As I explained in my [Hacking on Rosetta Tutorial](https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit?usp=sharing) it is common that video fails and crashes if you don't use a fan.
- If you still experience problems with the sample app, try to ask for support to DJI on their GitHub repo or forums.
- Consider replacing the video code mess in Rosetta with the code from the sample app following the guidelines I posted here on this issue. This would require creating a branch and removing **all traces** of old vídeo code. I was working on this until I discovered the video sample app didn't work on my model, ie. I can see the video on the screen but the streaming part (using Rosetta's NALU splitting code) is glitched. I believe the video format of my model is incompatible with the NALU splitting code.

kripper, 2023-04-07:
Regarding the video streaming, I'm suspicious about our NALU splitting code and the DJI video format, specially when I saw that DJI was using different "magic resource IDs" (depending on each model) to decode keyframes in the DJI video sample app.

Here someone made an important observation to our NALU splitting code:
https://github.com/roguestarslade/DJI-Android-VideoStreamDecodingSample/issues/1#issuecomment-1454732839

**Their team actually managed to receive the stream on Mavic 2 Pro and Mini 2.**

Anyway, the current video code in Rosetta (based on an older version of the DJI video sample app) must be replaced (this issue) because it also misses important things that were implemented in the DJI video sample app:
https://github.com/kripper/DJI-Android-VideoStreamDecodingSample

laurenweinstein1, 2023-04-07:
No luck building this test code, using the same Android Studio configuration that successfully builds Rosetta itself. Gets "no path to tools.jar for Android 31 API" error. Obvious fixes haven't worked so far.

kripper, 2023-04-07:
Sounds like a problem with your build environment (Android Studio).
But building Rosetta works for you?

laurenweinstein1, 2023-04-07:
Yeah. Both the main rosetta repo and the m4xw repo build fine. This appears to be a classic configuration problem that I can find people complaining about all over regarding Android Studio. Digging around ...

laurenweinstein1, 2023-04-08:
I think I'm going to stop trying to build this video streaming test code. Trying to make it build seems to risk confusing the configurations that permit me to successfully build the main Rosetta app and the m4xw versions.

So here's where it stands:

As I mentioned earlier, I can build the main app, get video in test mode, but no video in drone or simulator mode, and no streaming out of the video (e.g. to QGC).

Using the m4xw build, I can get (sometimes OK, sometimes messed up) video in all modes, and I really haven't experimented with trying to use that video feed (which often does work with QGC) for any AI apps. 

So I suspect I'm stuck until the video problem with the main app here for the mini 2 is solved.

kripper, 2023-04-08:
> Trying to make it build seems to risk confusing the configurations that permit me to successfully build the main Rosetta app and the m4xw versions

I use one same configuration for Rosetta and the DJI SDK Sample App.
I didn't need anything special to run the DJI SDK Sample App.

laurenweinstein1, 2023-04-08:
I think the problem is the different sets of JDKs and SDKs that I have installed. Once I found a combination that worked for building Rosetta I've been reluctant to change it. Attempts to upgrade Android Studio resulted in configurations that wouldn't build Rosetta. In the case of the Sample App, I get errors about no path to tools.jar for API 31. I do have Android 12/API 31 configured, but tools.jar doesn't exist anywhere on the system. In fact, apparently it doesn't exist at all for recent Android versions, so I don't know why the Sample is trying to use it, and I don't have an obvious way to fix it without risking the configuration that works for building Rosetta itself.

kripper, 2024-04-10:
The best way to go is implement https://github.com/RosettaDrone/rosettadrone/issues/175

closed 2024-04-10 by kripper