RosettaDrone/rosettadrone #132: EKF for vision landing

kripper, 2023-03-19:
I'm using visual pose estimation of Apriltags for precision landing (I'm preparing the code for releasing).

I see at least 3 problems to address:

1) how to get the attitude of the drone. a) Only use visual pose estimation, b) only use the drone positioning system or c) combine both.

If the error of the positioning system is too big, maybe it makes sense to only use vision.

d) We could also do our own position estimation based on the instructed motion (virtual stick actions).

2) the visual estimated pose will not be the current pose because of latency. This means, we would have to do corrections on the model using information from the past, to be able to use the model in the future (prediction). I believe EKF provides this solution.

3) which frame model to use? If the positioning system is accurate, a absolute frame model would be the best. Otherwise, if we use vision, maybe a frame relative to the Apriltag is convenient.

IMO an absolute positioning is more convenient because it allows us to compare positions among 3 or more objects (eg: the drone + two other objects).

kripper, 2023-03-20:
Here I published a Vision Landing implementation using AprilTags:
https://github.com/kripper/vision-landing-2

I also published a MAVLink Camera Simulator for testing.

chobitsfan, 2023-03-20:
> What is your experience with the latency drift (pose estimation is never current, so whatever motion instruction you send will always have some error).

I found latency matters more when drone is closer to landing target. This is a trade-off, high resolution image improve marker detect range but increase latency. I am thinking about a 2 stage approach, use high resolution image in the beginning, then switch to low resolution when we are closing to target

kripper, 2023-03-20:
I believe you are refering to the "maker detection latency", ie. the time to detect and estimate the pose of the big markers or the markers when they get close to the camera?

But I'm refering to the overall latency problem (caused by video capture, encoding, transmiting, decoding, detecting and estimating pose, sending mavlink commands, processing mavlink commands).

Imagine your drone takes a picture on `t = 0` while it is rotating the yaw and, based on that picture, determines that the landing point is to the left.
But at the time the motion command is executed on `t > 0` , the drone already rotated and the landing point is not on the left any more.
As a consequence, the drone will move in a wrong direction and be bouncing back and forth until reaching the target or crashing.

Of course, the lazy solution could be to move slowly or try to reduce the overall latency with better hardware...

But I believe that by using the correct model (EKF), we should be able to avoid moving in the wrong direction and get the drone landing fast and precise even with high latency.

Now, in your case, you just need to send the landing point to the flight controller so the EKF and the motion should be implemented there (in Ardupilot, PX4, RosettaDrone, et. al.).

In theory you just must make sure that you are telling the flight controller the exact timestamp of the image used for the pose estimation. And the flight controller should use this information **not to move into this direction**, but to adjust its estimation model parameters to predict the **future pose** of the landing point and to compute the direction where it should move to **in the present**.

But, it seems to me that the implementation (and probably also the MAVLink message definition) was or is still not doing it correctly.

@fnoop said: "There was a particular PR to solve a vital issue that never got implemented and I had been trying to get this resolved for so long that I basically gave up and moved on with life at that point..".
Here are more details: https://github.com/goodrobots/vision_landing/issues/123#issuecomment-1472896461

I'm researching...

danammeansbear, 2023-03-20:
if you can get access to the drone information like Lat,Long,Alt, gimbal angle and heading(north, south, east, west) in degrees then you can use geometry to guesstimate the approximate GPS location then heads towards that location. Create a PID controller based on your vision algorithm to use pose estimation for more precision.

kripper, 2023-03-22:
https://roboticsknowledgebase.com/wiki/common-platforms/dji-drone-breakdown-for-technical-projects/

kripper, 2023-03-22:
We will probably work with this implementation (WIP):
https://github.com/PX4/PX4-Autopilot/pull/20653

kripper, 2023-03-22:
I made some tests with a DJI Mini SE:
- `getVelocityX()` has only 1 digit precision, ie. it sends values like `0.1`, `0.2`, `0.3`,...10 cm precision is useless
- `getLatitude()` returns NaN when GPS signal is weak. Also useless.

Thus, we will have to work only with visual estimated pose and the sent virtual stick actions (to consider the visual delay).

kripper, 2023-03-29:
I finally came up with this algorithms:
https://github.com/kripper/SmartLanding

closed 2023-03-29 by kripper

kripper, 2023-04-08:
...but yaw information is reliable (and required):

[image]