RosettaDrone/rosettadrone #136: Mavlink messages with python script

Sasysan, 2023-03-24:
**Describe the bug**
Hi, i am trying to write a simple script that is a simple takeoff and go in a desired position but i am having some problem. I notice that with DroneKit and MAVSDK is not possible to arm and takeoff (also with qgorund when i try the command takeoff appear a pop-up that tell me that is not possible to arm the drone), while with pymavlink is possible to arm and takeoff. There is another problem when i set the height of takeoff for example 5 mt when i start the script the height is ignored and the drone go a the maxium RTL height that is set in the app.

kripper, 2023-03-24:
See: https://github.com/mavlink/MAVSDK/issues/1990

Feel free to add more messages to Rosetta to improve compatibility with more MAVLink implementations.

For the RTL altitude, check in the DJI manual how RTL works.

Sasysan, 2023-03-29:
The problem was not with the `MissionManager` but with `motion.mask.ignorePosZ = true;` in DroneModel.java line  1969 if you comment that line the drone will go to the desire altitude.

kripper, 2023-03-29:
Ok, this is new code and it probably broke recently. I will check and fix very soon.

closed 2023-03-29 by kripper