RosettaDrone/rosettadrone #137: Integrate CI with this repo and provide binary packages for users

pktiuk, 2023-03-29:
**Is your feature request related to a problem? Please describe.**

Rosetta Drone seems to be a very useful and nice app, but its adoption is limited mainly by lack of easy to obtain binary apk packages for end-users. Current user has to be able to compile this app and have account in DJI to get Application key and obtain Google Maps API key.   
I think it would be great to provide these packages in automated manner to avoid maintenance burden. Integrating build actions could also help with checking quality of code in Pull requests.

**Describe the solution you'd like**

- implement Build actions linked with pushing and PR-s
- Implement build an release actions

**Describe alternatives you've considered**

- building manually
- leaving it as it is

**Additional context**

In terms of safety we could use [GitHub Secrets](https://docs.github.com/en/actions/security-guides/encrypted-secrets) to ensure Application key won't leak.

pktiuk, 2023-03-29:
@The1only    
I would be interested in implementing these features, but firstly I want to get an approval.  And discuss it:)

kripper, 2023-03-29:
Hi @pktiuk ,

I think it is not a good idea to release binaries because,
- Rosetta is focused on developers, not final users
- there is risk involved. You should always test your model in a simulator first and also be prepared to disconnect the RC immediately if you see something is going wrong.
- you need a license key from DJI
- you need a license hey from Google

Dup of https://github.com/The1only/rosettadrone/issues/95

I believe GitHub is already checking if the project builds after a commit.

closed 2023-03-29 by kripper

pktiuk, 2023-03-29:
Hi @kripper ,   
I disagree with some of your points.

> Rosetta is focused on developers, not final users

Having builds would make testing this app and finding bugs much easier. Without builds people can't even open this app to check interface to check whether this app meets their needs.

> there is risk involved. You should always test your model in a simulator first and also be prepared to disconnect the RC immediately if you see something is going wrong.

To be honest I don't understand how this point is linked with releasing binaries.

> you need a license key from DJI

Why is this a problem? Key can be easily generated and AFAIK are no limits from DJI in usage of these keys. I think they serve similar purpose to tokens required by discord bots.

>  you need a license hey from Google

I guess token from google may have limited number of API calls, co we can ask user about generating it and saving in settings.

> I believe GitHub is already checking if the project builds after a commit.

No, this repository doesn't have any workflows, and has never ran any build actions. Just check [actions tab](https://github.com/The1only/rosettadrone/actions). This entire repository may not compile as well.

pktiuk, 2023-03-30:
> I'm afraid I'm the only active maintainer and I'm sure I will not give support to users who didn't build the project themselves.

It explains a lot, but without users this project won't get traction, and without traction it wouldn't get any outside contributions. It is also more difficult to find this project in the internet because nobody uses it. I spent 3 days looking for simple solution for better streaming from DJI drone without using official app and I found  it on some old forum briefly mentioning legacy version of this app.

pktiuk, 2023-03-30:
Anyway I think at least basic CI building this app would be very useful, so this issue should be either reopened, or I can create a new one, only about creating CI (only building without publishing releases).

kripper, 2023-03-30:
CI is welcome.

pktiuk, 2023-03-30:
PR is waiting https://github.com/The1only/rosettadrone/pull/140 , you have to only figure out what is wrong with your repo config