RosettaDrone/rosettadrone #139: App crashes when launched with Android Studio - dji.keysdk.callback.KeyListener not found

pktiuk, 2023-03-30:
### Describe the bug

I compiled app and wanted to open it to check its functionalities.

### Steps To Reproduce

1. Build app according to instructions in README
2. Create device in Android Studio
3. Press `Run 'app'`

### Desktop

```Markdown
- OS: Ubuntu
- Browser Firefox
- Version  22.04
```

### Smartphone

```Markdown
- Device: Nexus 7 (2012) and Nexus 9 (API 30)
- OS: Android 11
- Browser [e.g. stock browser, safari]
- Version: App compiled from master, commit 61f4a2a904edefc93d03ca253c0519e5f4597279
```

### Additional context

Full error:

```
$ adb shell am start -n "sq.rogue.rosettadrone/sq.rogue.rosettadrone.ConnectionActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 3371 on device 'Nexus_9_API_30 [emulator-5554]'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
I/ue.rosettadron: Late-enabling -Xcheck:jni
I/ue.rosettadron: Unquickening 12 vdex files!
W/ue.rosettadron: Unexpected CPU variant for X86 using defaults: x86
I/ndk_translation: Initialized NDK translation (armv7l), version 0.2.2
I/ue.rosettadron: The ClassLoaderContext is a special shared library.
I/chatty: uid=10121(sq.rogue.rosettadrone) identical 1 line
I/ue.rosettadron: The ClassLoaderContext is a special shared library.
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
W/GooglePlayServicesUtil: sq.rogue.rosettadrone requires the Google Play Store, but it is missing.
D/libEGL: loaded /vendor/lib/egl/libEGL_emulation.so
D/libEGL: loaded /vendor/lib/egl/libGLESv1_CM_emulation.so
D/libEGL: loaded /vendor/lib/egl/libGLESv2_emulation.so
D/AndroidRuntime: Shutting down VM
    
    
    --------- beginning of crash
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: sq.rogue.rosettadrone, PID: 3371
    java.lang.NoClassDefFoundError: Failed resolution of: Ldji/keysdk/callback/KeyListener;
        at java.lang.Class.newInstance(Native Method)
        at android.app.AppComponentFactory.instantiateActivity(AppComponentFactory.java:95)
        at androidx.core.app.CoreComponentFactory.instantiateActivity(CoreComponentFactory.java:45)
        at android.app.Instrumentation.newActivity(Instrumentation.java:1253)
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3353)
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3601)
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85)
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135)
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2066)
        at android.os.Handler.dispatchMessage(Handler.java:106)
        at android.os.Looper.loop(Looper.java:223)
        at android.app.ActivityThread.main(ActivityThread.java:7656)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "dji.keysdk.callback.KeyListener" on path: DexPathList[[zip file "/data/app/~~4zfmKB2FrrGtLwBloMvQlw==/sq.rogue.rosettadrone-yeHMSpGipR3vMVj9aMbgHA==/base.apk"],nativeLibraryDirectories=[/data/app/~~4zfmKB2FrrGtLwBloMvQlw==/sq.rogue.rosettadrone-yeHMSpGipR3vMVj9aMbgHA==/lib/arm, /data/app/~~4zfmKB2FrrGtLwBloMvQlw==/sq.rogue.rosettadrone-yeHMSpGipR3vMVj9aMbgHA==/base.apk!/lib/armeabi-v7a, /system/lib, /system_ext/lib]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:207)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at java.lang.Class.newInstance(Native Method) 
        at android.app.AppComponentFactory.instantiateActivity(AppComponentFactory.java:95) 
        at androidx.core.app.CoreComponentFactory.instantiateActivity(CoreComponentFactory.java:45) 
        at android.app.Instrumentation.newActivity(Instrumentation.java:1253) 
        at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:3353) 
        at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:3601) 
        at android.app.servertransaction.LaunchActivityItem.execute(LaunchActivityItem.java:85) 
        at android.app.servertransaction.TransactionExecutor.executeCallbacks(TransactionExecutor.java:135) 
        at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:95) 
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2066) 
        at android.os.Handler.dispatchMessage(Handler.java:106) 
        at android.os.Looper.loop(Looper.java:223) 
        at android.app.ActivityThread.main(ActivityThread.java:7656) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:592) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:947) 
I/Process: Sending signal. PID: 3371 SIG: 9

```

kripper, 2023-03-30:
Unexpected CPU variant for X86 using defaults: x86
You need to use a supported mobile device.
@The1only you got it working on X86?

The1only, 2023-03-30:
No I used Android Studio on Apple M1, there is works (Arm64)Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-03-30:
Should we remove the x86 architecture? Apple M1 seems to be ARM.
Any idea why we added it?

The1only, 2023-03-30:
I tried as DJI did use to support it many years ago, did not work even then, and no need any more. Remove it.Best regardTerje Nilsen9Tek ASNorway