RosettaDrone/rosettadrone #141: Build command failed- I can not build the app using Android Studio on Windows 7

OMAHDrone, 2023-03-30:
Describe the issue
I have downloaded the project and import it inside Android Studio but l could not build the app.
My machine specifications :
Operating system: 64-bit Windows 7
Processor: Intel i5
Physical phone that l physically connected to my laptop and built the app for: Redmi Note 9 Pro
Error message:
.....
[1/1] Linking C shared library libdjivideojni.so
FAILED: libdjivideojni.so 
cmd.exe /C "cd . && C:\Users\Robot\AppData\Local\Android\Sdk\ndk\22.0.7026061\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe --target=aarch64-none-linux-android22 --gcc-toolchain=C:/Users/R/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/R/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/arm64-v8a/libffmpeg.so -llog -latomic -lm && cd ."
ld: error: ../../../../src/main/cpp/lib/arm64-v8a/libffmpeg.so:1: unknown directive: version
>>> version https://git-lfs.github.com/spec/v1
>>> ^
clang: error: linker command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.

May l know what should I do to build the app for my phone?
Thank you very much for your help

kripper, 2023-03-30:
Today we transfered the repo here and then received this E-mail from Github:

```
Git LFS has been disabled on the organization RosettaDrone because you’ve exceeded your data plan by at least 150%. Please purchase additional data packs to cover your bandwidth and storage usage:

  https://github.com/organizations/RosettaDrone/billing/data/upgrade

Current usage as of 30 Mar 2023 07:36PM CEST:

  Bandwidth: 0.0 GB / 1 GB (0%)
  Storage: 2.5 GB / 1 GB (250%)
```

I hope it fixes automatically tomorrow.

kripper, 2023-03-30:
In the meanwhile, you can checkout my repo here:
https://github.com/kripper/rosettadrone

It's up to date with master.

kripper, 2023-03-30:
@The1only can we delete some or all branches? Which ones?

The folder `rosettadrone\app\src\main\cpp\lib` is using over `500 MB`.

kripper, 2023-03-30:
According to CI, my repo may also be broken.
Reported here: https://github.com/orgs/community/discussions/51525

The1only, 2023-03-31:
Erase is all, but remember I have a payd subscription to host Rosetta, allways did. Even if you clean up, there is a problem. 
If you do not then the video modules will be missing, it might be that removing x86 helps.

kripper, 2023-03-31:
The latest DJI video sample app (upstream) only uses some files:
https://github.com/RosettaDrone/DJI-Android-VideoStreamDecodingSample/tree/develop/android-videostreamdecodingsample/libs

@The1only did you need to compile ffmpeg yourself or could we maybe just replace with those files from upstream?

The1only, 2023-03-31:
The ffmpeg was inn and out a few times. 
I do not remember but was it Phantom4 og Inspire 1 that required it? It was something that required it at some point in time. 
How did I get the files? I might have built is from source at som point, I did the same with gstreamer. But it was so many things going on I cant exactly remember.

kripper, 2023-03-31:
Ok. Anyway, our best option is to stick to upstream's code (using a wrapper class) and send them a pull request in the strange case we needed to patch something.

pktiuk, 2023-04-01:
As far as I know, storing big binary files (especially libraries) in a git repository [is considered a bad practice](https://stackoverflow.com/questions/29674006/is-it-good-practice-to-store-binary-dependencies-in-source-control). 

There should be another way of managing these dependencies. In case of regular Linux app compilers just use system apps obtained with package managers.

The1only, 2023-04-01:
Mybe… I always used LFS, newer had an issue. I like collecting files in one place.  I do not belive DJI uses any of there files, they are for transcoding only. We can/could set output format independant of input format, important for post processing.

kripper, 2023-04-01:
The ffmpeg libraries are indeed used by the video sample app, and they already reduced the size of these files from 560 MB to like 10 MB (they probably removed unused files).

OMAHDrone, 2023-04-01:
Dear rosettadrone developers,
May you help me because l could not build the app. Ineed the app for my school project using the dji mini 2. I tried for two days but l could not build the app.

My machine specifications :
Operating system: 64-bit Windows 7
Processor: Intel i5
Physical phone that l physically connected to my laptop and built the app for: Redmi Note 9 Pro
Your help and support is highly appreciated.
Th error message:

Task :app:externalNativeBuildDebug FAILED
Execution failed for task ':app:externalNativeBuildDebug'.
> Build command failed.
  Error while executing process C:\Users\Robot\AppData\Local\Android\Sdk\cmake\3.10.2.4988404\bin\ninja.exe with arguments {-C C:\Users\Robot\rosettadrone\app\.cxx\cmake\debug\armeabi-v7a djivideojni}
  ninja: Entering directory `C:\Users\Robot\rosettadrone\app\.cxx\cmake\debug\armeabi-v7a'
  [1/2] Building C object CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o
  In file included from ../../../../src/main/cpp/dji_video_jni.c:4:
  In file included from ../../../../src/main/cpp\ffmpeg.h:34:
  In file included from ../../../../src/main/cpp/cmdutils.h:28:
  In file included from ../../../../src/main/cpp/include\libavcodec/avcodec.h:31:
  In file included from ../../../../src/main/cpp/include\libavutil/samplefmt.h:24:
  In file included from ../../../../src/main/cpp/include\libavutil/avutil.h:289:
  In file included from ../../../../src/main/cpp/include\libavutil/common.h:401:
  In file included from ../../../../src/main/cpp/include\libavutil/mem.h:33:
  ../../../../src/main/cpp/include\libavutil/error.h:121:5: warning: non-constant static local variable in inline function may be different in different files [-Wstatic-local-in-inline]
      static char str[AV_ERROR_MAX_STRING_SIZE];
      ^
  ../../../../src/main/cpp/include\libavutil/error.h:119:1: note: use 'static' to give inline function 'av_err2str' internal linkage
  av_always_inline char *av_err2str(int errnum) {
  ^
  static 
  ../../../../src/main/cpp/include\libavutil/attributes.h:37:30: note: expanded from macro 'av_always_inline'
  #    define av_always_inline __attribute__((always_inline)) inline
                               ^
  1 warning generated.
  [2/2] Linking C shared library libdjivideojni.so
  FAILED: libdjivideojni.so 
  cmd.exe /C "cd . && C:\Users\Robot\AppData\Local\Android\Sdk\ndk\22.0.7026061\toolchains\llvm\prebuilt\windows-x86_64\bin\clang.exe --target=armv7-none-linux-androideabi22 --gcc-toolchain=C:/Users/Robot/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64 --sysroot=C:/Users/Robot/AppData/Local/Android/Sdk/ndk/22.0.7026061/toolchains/llvm/prebuilt/windows-x86_64/sysroot -fPIC -g -DANDROID -fdata-sections -ffunction-sections -funwind-tables -fstack-protector-strong -no-canonical-prefixes -D_FORTIFY_SOURCE=2 -march=armv7-a -mthumb -Wformat -Werror=format-security  -O0 -fno-limit-debug-info  -Wl,--exclude-libs,libgcc.a -Wl,--exclude-libs,libgcc_real.a -Wl,--exclude-libs,libatomic.a -static-libstdc++ -Wl,--build-id=sha1 -Wl,--no-rosegment -Wl,--fatal-warnings -Wl,--exclude-libs,libunwind.a -Wl,--no-undefined -Qunused-arguments -shared -Wl,-soname,libdjivideojni.so -o libdjivideojni.so CMakeFiles/djivideojni.dir/src/main/cpp/dji_video_jni.c.o  ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so -llog -latomic -lm && cd ."
  ld: error: ../../../../src/main/cpp/lib/armeabi-v7a/libffmpeg.so:1: unknown directive: version
  >>> version https://git-lfs.github.com/spec/v1
  >>> ^
  clang: error: linker command failed with exit code 1 (use -v to see invocation)
  ninja: build stopped: subcommand failed.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

[image]
[image]
[image]

pktiuk, 2023-04-01:
@OMAHDrone     
I guess you can try to download repository as ZIP and that file should contain these missing library files.
[image]

Just unzip that file and paste missing files.

OMAHDrone, 2023-04-01:
Thank you very much. I will try now and update you.
Thank you very much

kripper, 2023-04-01:
You just need to copy `libffmpeg.so` inside `src\main\cpp\lib\arm64-v8a`

kripper, 2023-04-01:
I created an issue for Github here: https://support.github.com/ticket/personal/0/2088957

OMAHDrone, 2023-04-02:
Thank you very much kripper for your help.
May l have the link to download the file libffmpeg.so?
Thank you very much

kripper, 2023-04-03:
> Thank you very much kripper for your help. May l have the link to download the file libffmpeg.so? Thank you very much

https://github.com/RosettaDrone/DJI-Android-VideoStreamDecodingSample/tree/develop/android-videostreamdecodingsample/libs

pktiuk, 2023-04-03:
Just in case someone needs to download them quickly:   [libffmpeg.zip](https://github.com/RosettaDrone/rosettadrone/files/11136011/libffmpeg.zip)

kripper, 2023-04-03:
LFS files have been purged.
I also re-added the required binary files.
Build should work again.

closed 2023-04-03 by kripper

OMAHDrone, 2023-04-04:
Thank you very much kripper  and pktiuk for libffmpeg files. Now the app can compile and build for dji Mini 2. However, l can not control the dji mini 2 using the app although l receive the video feed and telemetry info.
In terms of QCS using windows, I can only receive the telemetry data but l can not receive the video feed. The screen sometimes becomes green. I also can not takeoff using GCS.
May l know what is the problem? I need to use this app for AI python code development for my school project.
Thank you very much for your help and time.
My pleasure

kripper, 2023-04-05:
Please check comments in other issues and discussions.

KroSmi, 2024-01-19:
i get the same error the libraries aren't missing and i have git lfs installed 
[image]