RosettaDrone/rosettadrone #145: Rosetta Drone from master branch crashes on DJI Smart Controller

pktiuk, 2023-04-03:
### Describe the bug

I wanted to test this app with My DJI Smart Controller Enterprise but enabling TestMode crashed it.

### Steps To Reproduce

Build
1. Clone master branch
2. Build it using instructions from README
3. Apply change from https://github.com/RosettaDrone/rosettadrone/pull/143 (without this there are errors)
4. Build APK file 
5. Upload and install it to DJI Smart Controller

Run:
1. Open app
2. press logo several times to enable TestMode
3. App crashes 'Rosetta Drone has stopped`

### Desktop

```Markdown
- OS: Ubuntu 22.04
```

### Smartphone

```Markdown
- Device: DJI Smart Controller Enterprise
- OS: Android
- Version 7.1.2
```

### Additional context

When I opened and compiled app from `work` branch (with applied https://github.com/RosettaDrone/rosettadrone/pull/143) app does not crash.

pktiuk, 2023-04-03:
Is it linked with https://github.com/RosettaDrone/rosettadrone/issues/124 ?

kripper, 2023-04-03:
> Is it linked with #124 ?

These changes have already been merged and I believe DJI Smart Controller was working for @The1only.

The1only, 2023-04-03:
Yes I uses DJI smart controller and it is working fine (but I have not done a git pull in a few weeks.)

kripper, 2023-04-03:
@pktiuk please pull recent changes and send a stacktrace or log if you can.
I don't have a DJI Smart Controller, but maybe @The1only can tell you how to debug or get the logs.
I would try to start wifi adb on the device, maybe installing an app for that.
Another option would be to add some component to capture the stacktraces inside RosettaDrone (eg. ACRA).

pktiuk, 2023-04-04:
I guess, I cannot debug Enterprise edition because of this bottom part of the device is some kind of HUB blocking usb port.
[image]
I will try to remove it and then test it again with connected ADB.

kripper, 2023-04-04:
Did you try installing an app to enable ADB via Wifi (without using an USB port)?

pktiuk, 2023-04-04:
Android 7 is too low version for debugging via WiFI

pktiuk, 2023-04-05:
I got 
```log
2023-04-05 09:45:46.537   225-438   BufferQueueProducer     pid-225                              E  [Waiting For Debugger: sq.rogue.rosettadrone] connect: already connected (cur=1 req=1)
2023-04-05 09:46:08.756   739-739   ResourcesManager        pid-739                              E  failed to add asset path /data/app/sq.rogue.rosettadrone-2/base.apk
2023-04-05 09:46:08.756   739-739   ResourcesManager        pid-739                              E  failed to add asset path /data/app/sq.rogue.rosettadrone-2/base.apk
2023-04-05 09:46:08.756   739-739   ResourcesManager        pid-739                              E  failed to add asset path /data/app/sq.rogue.rosettadrone-2/base.apk
2023-04-05 09:46:09.558  2373-2373  DJISDKManager           pid-2373                             E  Failed to load LDM Plus license file /storage/emulated/0/DJI/sq.rogue.rosettadrone.dlf (No such file or directory)
2023-04-05 09:46:09.705   225-438   BufferQueueProducer     pid-225                              E  [sq.rogue.rosettadrone/sq.rogue.rosettadrone.ConnectionActivity] connect: already connected (cur=1 req=1)
2023-04-05 09:46:27.949  2373-2373  DJISDKManager           pid-2373                             E  Failed to load LDM Plus license file /storage/emulated/0/DJI/sq.rogue.rosettadrone.dlf (No such file or directory)
--------- beginning of crash
2023-04-05 09:46:28.222  2836-2836  DEBUG                   pid-2836                             A  pid: 2373, tid: 2680, name: async_parser  >>> sq.rogue.rosettadrone <<<
2023-04-05 09:46:28.224  2836-2836  DEBUG                   pid-2836                             A      #00 pc 0000000000001a50  /data/app/sq.rogue.rosettadrone-1/lib/arm64/libdjivideojni.so (noparse+80)
2023-04-05 09:46:28.224  2836-2836  DEBUG                   pid-2836                             A      #01 pc 0000000000001b10  /data/app/sq.rogue.rosettadrone-1/lib/arm64/libdjivideojni.so (Java_sq_rogue_rosettadrone_video_NativeHelper_parse+140)
2023-04-05 09:46:28.224  2836-2836  DEBUG                   pid-2836                             A      #02 pc 0000000001c03ccc  /data/app/sq.rogue.rosettadrone-1/oat/arm64/base.odex (offset 0x1ace000)
2023-04-05 09:46:30.438   225-235   BufferQueueProducer     pid-225                              E  [Application Error: sq.rogue.rosettadrone] connect: already connected (cur=1 req=1)
2023-04-05 09:46:30.439   483-671   InputDispatcher         pid-483                              E  channel 'fc12f08 sq.rogue.rosettadrone/sq.rogue.rosettadrone.MainActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
2023-04-05 09:46:30.465   483-671   InputDispatcher         pid-483                              E  channel '6448c39 sq.rogue.rosettadrone/sq.rogue.rosettadrone.ConnectionActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
2023-04-05 09:46:36.637  2918-2918  art                     sq.rogue.rosettadrone                I  Late-enabling -Xcheck:jni
2023-04-05 09:46:36.648  2918-2925  art                     sq.rogue.rosettadrone                I  Ignoring second debugger -- accepting and dropping
2023-04-05 09:46:36.649  2918-2925  art                     sq.rogue.rosettadrone                I  Ignoring second debugger -- accepting and dropping
2023-04-05 09:46:36.938  2918-2918  DjiCoreHelper           sq.rogue.rosettadrone                D  mask=
2023-04-05 09:46:36.951  2918-2918  GooglePlayServicesUtil  sq.rogue.rosettadrone                W  sq.rogue.rosettadrone requires the Google Play Store, but it is missing.
2023-04-05 09:46:36.969  2918-2918  VideoService            sq.rogue.rosettadrone                E  oncreate Video 
2023-04-05 09:46:36.979  2918-2918  NativeHelper            sq.rogue.rosettadrone                E  NativeHelper
2023-04-05 09:46:36.985  2918-2918  DJI                     sq.rogue.rosettadrone                D  CDecoder::prepare()2
```
[rosetta_crash.log](https://github.com/RosettaDrone/rosettadrone/files/11156394/rosetta_crash.log)
[rosetta_no_crash_after_restart.log](https://github.com/RosettaDrone/rosettadrone/files/11156396/rosetta_no_crash_after_restart.log)

Sometimes after restart the app does not crash.

pktiuk, 2023-04-05:
Debugging gives me this result:   
[image]

[debugging_attempt.log](https://github.com/RosettaDrone/rosettadrone/files/11156822/debugging_attempt.log)

pktiuk, 2023-04-05:
I got also logs from `/sdcard/dji_logs` 
[Compressed_logs.zip](https://github.com/RosettaDrone/rosettadrone/files/11158140/Compressed_logs.zip)

pktiuk, 2023-04-05:
> Did you check https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/mobilebasic ?

I forgot, My fault. :facepalm:

pktiuk, 2023-04-05:
Now I get in Console:
```
E/WP_JNI: JNI_OnLoad
I/WP_JNI: registerDevice in jni :63
E/WP_JNI: updateDeviceState in jni
E/WP_JNI: updateDeviceState in jni
E/WP_JNI: updateDeviceState in jni
I/WP_JNI: current device id in deviceSupportWPV2Handler:63
I/WP_JNI: current device id in isDeviceSupportSDRHandle:63
D/MainActivity: setDroneParameters
D/MainActivity: Starting Video link to 192.168.12.1:5600
D/MainActivity: MATRICE_300_RTK Connected 
D/MainActivity: Do NOT Transcode Video !!!!!!!
D/MainActivity: Fetched param from DJI: DJI_FAILSAFE=GO_HOME
I/WP_JNI: current device id in deviceSupportWPV2Handler:63
I/WP_JNI: current device id in deviceSupportWPV2Handler:63
E/WP_JNI: setV1WaypointStateChangeCallback Success:
E/NativeHelper: NativeHelper
A/libc: Fatal signal 11 (SIGSEGV), code 1, fault addr 0x148 in tid 3906 (async_parser)
E/WP_JNI: current string length 4
W/System.err: java.io.FileNotFoundException: /proc/-1/oom_adj (No such file or directory)
W/System.err:     at java.io.FileInputStream.open(Native Method)
W/System.err:     at java.io.FileInputStream.<init>(FileInputStream.java:146)
W/System.err:     at java.io.FileInputStream.<init>(FileInputStream.java:99)
W/System.err:     at java.io.FileReader.<init>(FileReader.java:58)
W/System.err:     at dji.midware.util.ContextUtil.isAppBroughtToBackground(Unknown Source)
W/System.err:     at dji.midware.xss.nhf.gfd.nhf(Unknown Source)
W/System.err:     at dji.midware.sockets.pub.nhf$fdd.run(Unknown Source)
W/System.err:     at java.util.TimerThread.mainLoop(Timer.java:555)
W/System.err:     at java.util.TimerThread.run(Timer.java:505)
Disconnected from the target VM, address: 'localhost:37757', transport: 'socket'
```

kripper, 2023-04-05:
Probably related with the video binaries.

Are you trying the sample app?
https://github.com/RosettaDrone/rosettadrone/issues/131

pktiuk, 2023-04-06:
I tried the sample app (report in comment: https://github.com/RosettaDrone/rosettadrone/issues/131#issuecomment-1498674523 )

That sample app does not crash. I hope that report helps.

closed 2023-04-06 by kripper