RosettaDrone/rosettadrone #149: Dji Mini2, rosettadrone, QGC

OMAHDrone, 2023-04-05:
Dear rosettadrone developers, I have just downloaded the new updated app cloned from https://github.com/RosettaDrone/rosettadrone and the app was built successfully. I have the following issue:
1- I can not open the app in Drone or Simulator mode from the first time run. I need first to open the Test mode then l can go back and change the mode.
2 - Now I can receive the video feed  from my dji mini 2 to rosettadrone 3 app to QGC software in Test mode only. I can not receive the video feed  on QGC software when the app is in Drone mode.
3- Now l can takeoff from rosettadrone 3 app when the app is in Drone mode. I still can not takeoff using QGC software. When l land the drone using the QGC software, the drone slowly decreases altitude and then hovers. I can not land the drone even if I use the rosettadrone 3 app.
May you help me. I also want to know how to control drone using drone images directly from python dronekit instead of QGC software.
Thank you very much for your help
My pleasure

kripper, 2023-04-05:
> 1- I can not open the app in Drone or Simulator mode from the first time run. I need first to open the Test mode then l can go back and change the mode.

I tested today and it connects fine for me.
Pull latest version and put a breakpoint on `refreshProductInfo()` to see what's going on.

> 2 - Now I can receive the video feed from my dji mini 2 to rosettadrone 3 app to QGC software in Test mode only. I can not receive the video feed on QGC software when the app is in Drone mode.

Are you sure you can see the drone camera in QGC using Test Mode?

> 3- Now l can takeoff from rosettadrone 3 app when the app is in Drone mode. I still can not takeoff using QGC software. When l land the drone using the QGC software, the drone slowly decreases altitude and then hovers. I can not land the drone even if I use the rosettadrone 3 app.

I will test the landing code soon again.

> May you help me. I also want to know how to control drone using drone images directly from python dronekit instead of QGC software.

Here again:
https://github.com/RosettaDrone/rosettadrone/discussions/133#discussioncomment-5383376

OMAHDrone, 2023-04-07:
Thank you very much kripper.
I will reclone the app and test again and then let you know
Thank you very much for your help
My pleasure

kripper, 2023-04-07:
I'm closing this issue, because it's creating a redundant discussion.
In the future, 1) post one issue per issue and 2) comment on existing issues.

closed 2023-04-07 by kripper