RosettaDrone/rosettadrone #150: Video only in Test mode on Mini 2

laurenweinstein1, 2023-04-07:
### Describe the bug

I've been working with RD for a long time, mostly with the m4xw forks (in coordination with the forks' owner) since until the Mini 2 had native MSDK support, I needed the emulation that the m4xw build provided.

I just built the current version over here. It seems to be basically functioning, except I'm seeing no video on the Mini 2 (only black) on RD itself except in Test mode. Simulator and Drone mode are black, Test mode comes right up with the video from the drone. Settings seem correct and normal.

Any ideas? 

I should add that I'm particularly interested in precision landing possibilities and other AI applications. I also have a Mini 3 Pro which hopefully will be supported by the MSDK very shortly Dronelink appears to already have a beta of a new MSDK 5 version with this support -- and BTW they report that DJI has disabled obstacle avoidance in it, unlike in MSDK 4 where it functioned. Dronelink has asked them re-enable this and is hopeful that they will at some point.

Thanks!

### Desktop

```Markdown
- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]
```

### Smartphone

```Markdown
- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]
```

OMAHDrone, 2023-04-07:
Dear laurenweinstein1, I have the same problem with my dji mini 2 and l have already reported it.
Did you manage to control dji mini 2 and receive video feed in python using dronekit. If yes, may you help me with the code. This is because l have an AI project using the dji mini 2 but l could not manage to control the drone or receive its video feed using dronekit
Thank you very much again for your help
My pleasure

laurenweinstein1, 2023-04-07:
Hi. I just did the build this afternoon, and wasn't planning to try any actual flights until the video situation is understood. I won't be trying any AI-related (e.g. dronekit) actions that depend on the video until the basic video feed is functional in the app.

OMAHDrone, 2023-04-07:
Thank you very much laurenweinstein1 for the clarification. I will be very grateful to you to help me with the minimum dronekit skeleton code that will allow me to receive the dji mini 2 video feed and do basic control for forward/backward, leftward/rightward, upward/downward, and yaw motions. I will then add my AI logic to complete the code. I will be waiting for any updates from you as l have to submit my project in June and l do not have much time.
Thank you very much again for your help, time, and support
My pleasure

kripper, 2023-04-07:
Hi,

I will be sponsoring and helping you both with your AI projects.
If video works in test mode, it can be easily fixed in drone and simulator mode.
But the problem is not seeing the video on Rosetta, but streaming it to Gstreamer (to QGC, DroneKit, etc).
@The1only got it working for some models, but not for the mini series.
To fix the streaming problem, see:
https://github.com/RosettaDrone/rosettadrone/issues/131

OMAHDrone, 2023-04-07:
Thank you very much kripperfor for helping us. We will make our projects open source. Our problem is that we do not have much experience with low-level programming stuff but we will do our best.  What we want is a dronekit python code that will enable us to receive the video stream from dji mini 2 and then do basic motion control. We can also pay you to quickly solve these problems in one day and make it open source. This is because we do not have much time as we have to quickly develop the AI agent and submit the project in June. Please give us step by step instructions on what we should do. We can also meet with you on zoom to quickly fix these problems.
We are waiting for your help and the time you want us to be available to quickly fix these problems.
We look forward to your reply
My pleasure

kripper, 2023-04-07:
Hi @OMAHDrone, don't worry. I guarantee you you will be processing the video input this Monday and start working with DroneKit.
And I don't want to be paid, thanks btw. My goal is to have you contribute to our Rosetta's AI ecosystem, document and share your work one it is ready and help us to attract more smart people to our opensource community.

OMAHDrone, 2023-04-07:
Thank you very much kripper for your help and support
We are with you until the end of this development and waiting for any instructions from you.
Thank you very much for your help
My pleasure

kripper, 2023-04-08:
@laurenweinstein1:

Here are the instructions (draft):
https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit#heading=h.q9mp6yjsfxe5

laurenweinstein1, 2023-04-08:
Thanks so much for this! I'm unsure if I understand the functionality of the video plugin that can be enabled. Does it currently work for the Mini 2, and (if so) can it be used to feed video for AI (e.g. dronelink) and also for QGC, etc.? Tnx again.

kripper, 2023-04-08:
Yes. Is it not clear? How can I improve the document?

laurenweinstein1, 2023-04-08:
My uncertainty on this point was caused by our earlier discussion about problems with video streaming from the Mini 2 (which implied to me that the DJI sample code was still being used to debug this problem). So if the plugin works with the Mini 2, it might be useful to explicitly list the specific drone models that will stream video successfully via the enabled plugin. Again, thanks for that great doc!

kripper, 2023-04-08:
Ok, I added a clarification for you.

closed 2023-04-08 by kripper

laurenweinstein1, 2023-04-08:
Thanks. By the way, it appears that aruco 3.1.0 is no longer available at the designated repo -- the latest I could find is 3.0.13.

kripper, 2023-04-08:
It's all clearly documented in the instructions.

laurenweinstein1, 2023-04-08:
The designated checkout fails because 3.1.0 is no longer apparently there. A note says that the 3.0.x series was the last completely open source.

kripper, 2023-04-08:
https://github.com/RosettaDrone/vision-landing-2/

laurenweinstein1, 2023-04-08:
Right, those instructions say to checkout 3.1.0. That will fail since it's no longer apparently in the repository. The latest you can get is 3.0.13.

kripper, 2023-04-08:
On my browser, I read this:

Aruco (>3.0.0) It is recommended to install 3.1.0 from https://github.com/fnoop/aruco as it has some fixes that haven't been released yet.
Installation is straight forward.

laurenweinstein1, 2023-04-08:
If you can find the 3.1.0 branch there please point me at it! I can't find it.

kripper, 2023-04-08:
https://github.com/fnoop/aruco

laurenweinstein1, 2023-04-08:
[image]
[image]

kripper, 2023-04-08:
Did you try this repo just in case?
Maybe he is referring to a "branch" from upstream and not necessarily replicated as a branch on his GitHub repo?

Anyway, CTRL + F also gave me this:

[image]

laurenweinstein1, 2023-04-08:
Yeah. OK. So it appears that what happened is that the licensing changed after 3.1.0, and only 3.0.x versions were left in the original repo. So the checkout instruction in the README fails since 3.1.0 is no longer available as a repo branch. The zip file does still exist though as you note, and I'm testing installing that now. Thanks.

kripper, 2023-04-08:
Maybe master is 3.1.0 + fixes, but he didn't update the README?
Otherwise, how did he generate and tag this 3.1.0 release?

I'm constantly updating my doc to make things easier.

By the way, what is the goal of your current project and what is your expertise?

laurenweinstein1, 2023-04-08:
3.1.0 built and installed fine. Thanks again.

My primary interest in this context is the degree to which commodity consumer equipment (computing, ground and aerial mobile devices, etc.) can potentially automate a range of tasks. Of course when it comes to consumer UAVs, precision landing is an absolutely foundational requirement, and to the extent it can be accomplished without requiring landing beacons and the like the issues become much more practicable. 

I go back all the way to the early days of AI and associated projects, when I was at UCLA in coordination with the Stanford and MIT AI labs starting in early ARPANET/Internet days.