RosettaDrone/rosettadrone #151: YUV buffer has 2048 additional bytes on Mini 2

kripper, 2023-04-10:
### Describe the bug

Moved from discussion.

Try: https://github.com/RosettaDrone/rosettadrone/commit/3af7161fabbfcc71afa5cc01616d850611dff8e5

### Desktop

```Markdown
.
```

### Smartphone

```Markdown
.
```

OMAHDrone, 2023-04-10:
Thank you veru much kripper
The problem is still there:
    yuv_mat = yuv_mat.reshape((h + h // 2, w))
ValueError: cannot reshape array of size 1384448 into shape (1080,1280)

OMAHDrone, 2023-04-11:
Thank you very much kripper
I have tested the app and code with dji air 2 and dji phantom 4. The high latency is still there (around 30 seconds at 30 fps). I noticed that dji phantom 4 needs the 2048 hack but dji air 2 doest not.
Thank you very much for your help
My pleasure

kripper, 2023-04-11:
I did some tests with my VisionLanding lab and got less than 1 [s] total latency (action + perception).

kripper, 2023-04-11:
Please create a different issue for latency.

Ugzuzg, 2023-07-02:
Do you still need help with this? I have a Mini 2 and can run thet test.

kripper, 2023-07-02:
More than testing, you could try fixing the mini 2 video compatibility problem by checking this repo:
https://github.com/m4xw/rosettadrone_mini2