RosettaDrone/rosettadrone #152: Drone is not detected

kripper, 2023-04-10:
### Describe the bug

Moved from discussión.

### Desktop

```Markdown
.
```

### Smartphone

```Markdown
.
```

OMAHDrone, 2023-04-10:
Thank you very much kripper
I have downloaded the repository again
I can confirm now I can open the Drone mode from the first time and it works fine
Thank you very much for your help
My pleasure

kripper, 2023-04-10:
But nothing important changed in the code regarding this isse, so the problem is probably still there, so you will have to figure out yourself how to make it work again, and let us know how you did it work, so we can maybe fix it (comment).

closed 2023-04-10 by kripper