RosettaDrone/rosettadrone #167: BUG: Missing brace in MAVLinkReceiver.java:190

marcomacchia99, 2023-06-01:
### Describe the bug

In `MavLinkReceiver.java`, starting from `line 187` it is written:

```java
case MAV_CMD_COMPONENT_ARM_DISARM:
     if (msg_cmd.param1 == 1)
          sendResponse(MAV_CMD_COMPONENT_ARM_DISARM, mModel.armMotors());
     else
          sendResponse(MAV_CMD_COMPONENT_ARM_DISARM);
          mModel.disarmMotors(true);
     break;
``` 

The incorrect code formatting doesn't clearly show the error, which is a missing brace for the else statement!
In this way, everytime this message is received, the motors are disarmed no matter the value of `param1`

marcomacchia99, 2023-06-01:
Moreover, the function `armMotors()` is fully commented. Uncommenting it results in the ability of arming a DJI Mini2 drone from QGC

kripper, 2023-06-01:
> The incorrect code formatting doesn't clearly show the error, which is a missing brace for the else statement! In this way, everytime this message is received, the motors are disarmed no matter the value of `param1`

That is clearly a bug. Fixed.

kripper, 2023-06-01:
> Moreover, the function `armMotors()` is fully commented. Uncommenting it results in the ability of arming a DJI Mini2 drone from QGC

Are you sure `mFlightController.turnOnMotors()` is needed for the DJI MIni 2 to take off?
Please try first without uncommenting this code, because I believe your problem was the other bug (missing brackets).

closed 2023-06-01 by kripper

marcomacchia99, 2023-06-02:
> Are you sure `mFlightController.turnOnMotors()` is needed for the DJI MIni 2 to take off? Please try first without uncommenting this code, because I believe your problem was the other bug (missing brackets).

`mFlightController.turnOnMotors()` is needed only to formally arm the DJI Mini 2 and start the motors, which I think is what the arm button of QGC should do

kripper, 2023-06-02:
I'm not sure why it was commented (probably because of safety reasons), but in my experience, taking off works fine without it.

I believe it is not necessary for DJI drones because the SDK starts the motors automatically when required.