RosettaDrone/rosettadrone #172: Problem building Rosetta Drone on Android Studio

davidan208, 2023-08-03:
Hi guys,
I'm following the setup instruction to build the rosetta drone for my DJI. I have initialized with git lfs install, generated key for Google Map API and DJI App and put it in the desired locations. But when I built on Android Studio 3.6.4 ( Gradle version 6.8.3), I encountered this 
[image]
Can any one give me a guide to overcome this problem, thanks a lot.

davidan208, 2023-08-03:
Hi @kripper 
Yes I have. I''ve tried many times but it ends up the same.

> Did you import the project using `File->New->Import Project` ?

kripper, 2023-08-03:
Maybe something changed in the cloud.
Please send a PR if you find the problem.

davidan208, 2023-08-03:
> Maybe something changed in the cloud. Please send a PR if you find the problem.

Thank you @kripper.
One question, when I install Android Studio, are there anything I should consider, or just doing as the recommend settings; and then just import this repository and build the apk without any additional steps (beside adding the keys.xml)

kripper, 2023-08-03:
No. I believe the correct grade version is automatically downloaded and installed when you import and build the project.

jashshah999, 2023-08-11:
Hi. Yes you can change your Android SDK Version. Go to File > Project Structure and under this you should see Android SDK Version which must be set to 17.x for you now. Click on the drop down and change it to 15. Then build the APK and you should be good to go!

davidan208, 2023-08-11:
@jashshah999 you're live saver. Thanks a lot my brother

kripper, 2023-08-11:
Did something change and should we update the README build instructions?

davidan208, 2023-08-11:
well actually it is about the SDK version, retry with SDK version 15 will immidiately ask for retry to build in Android Studio, and than just wait until done, cool project anyway @kripper

estevezb, 2023-08-19:
Hi, I am having the same issue but when i go to File > Project Structure i don't know where to find the Android SDK Version. 
[image]
[image]

Would it be possible to provide a screen shot of those fields and menu?

jashshah999, 2023-08-19:
Yes now click on grade settings. SDK version is now in Gradle settings.
You'll see the options there. Just click on the last link in your second
screenshot.

estevezb, 2023-08-19:
@jashshah999 . Please excuse my ignorance I am new to Android studio and Java. I clicked on the Gradle Settings link. Here is the new window. It is currently showing 17.02 which I expected based on your information. Do i download a new JDK version? You mentioned choosing 15 from this dropdown? If so, which one?, there are also multiple vendors listed but not sure that is relevant. Here is another screen shot to make sure i am on the same page: 
[image]

estevezb, 2023-08-19:
@jashshah999. NM. This worked!!! Thank you!!!

estevezb, 2023-08-19:
FYI, to the team, Readme could include , the specific vendor needed is azul 15  Azul Zulu : 
[image]

dentava, 2024-01-17:
Good morning, i reached the last step but it's not working for me. @estevezb when i build it returns a popup saying: Unable to find Gradle tasks to build: []. Build mode: ASSEMBLE. Tests: None. How did you solved?

broccan, 2024-06-04:
The position of gradle settings have now changed and can't access it via the SDK structure menu, instead you can find it using the following:
[image]  

[image]

CAMIprojectsUNI, 2024-08-17:
I have encountered the same errors as previous users. I followed their steps and their help was useful up to a certain point... but I get this error when trying to Build->Make Project.
[image]

angel-ayala, 2024-09-05:
Apparently seems to be a version problem with a dependency from Mapbox v9 which by the way [is not supported anymore](https://github.com/mapbox/mapbox-maps-android/issues/2457#issuecomment-2293256647)., I don't know exactly if is still being served since it continues appearing in [maven's repo](https://mvnrepository.com/artifact/com.mapbox.mapboxsdk/mapbox-android-accounts/0.7.0)

HEEJUN-SHIN, 2024-09-09:
Hello, @angel-ayala @CAMIprojectsUNI, the problem comes from  DJI UXSDK 4:16 and 
I found temporal solution from DJI Developer support.

https://sdk-forum.dji.net/hc/en-us/articles/37229365477657-During-the-compilation-of-the-MSDK-V5-example-where-it-stated-Could-not-find-com-mapbox-mapboxsdk-mapbox-android-accounts-0-7-0-How-should-address-this

you can add
```
 maven { url 'https://maven.aliyun.com/repository/public' }
 maven { url 'https://maven.aliyun.com/repository/google' }
 maven { url 'https://repo.huaweicloud.com/repository/maven'}
```

inside allprojects/repositories in project-level build.gradle.

angel-ayala, 2025-02-20:
@ricricricricric  I'm not sure about what the problem is, however, appears that you are working on a labeled sourced code, have you tried making a clone of the repo? or download the master branch?

Reto80, 2025-08-28:
Hallo und Guten Tag, danke vielmals für Ihre Arbeit (RosettaDrone), jedoch scheitere ich am erstellen der App und brauche Ihre Hilfe Arbeite momentan mit Android Studio 2021.1.3.1 mit Azul Zulu 15.0.10 Android Gradle Plugin Version 3.6.4 und Gradle Version 7.5 natürlich habe ich die Key Datei mit DJI und Gogle Api Key erstellt. I need Help  from Your Smart Brain. Best Wishes from Swiss