RosettaDrone/rosettadrone #174: Getting stream on QGC but not using gstreamer

jashshah999, 2023-08-11:
### Describe the bug

Hello. I am getting my rosettadrone feed on QGC well using a UDP H.264 encoding but when I try to do the same using just gstreamer, the stream is extremely laggy and loses packets. 

### Steps To Reproduce

1. Run QGC with the correct port and see the stream correctly. 
2. Run the same port with gstreamer and see the buggy/laggy stream 

### Screenshots

[Screencast from 08-10-2023 10:31:39 PM.webm](https://github.com/RosettaDrone/rosettadrone/assets/49280550/1791dec8-8cbd-4fd4-97d2-a306cd463293)
[Screencast from 08-10-2023 10:32:13 PM.webm](https://github.com/RosettaDrone/rosettadrone/assets/49280550/80267953-4da6-45cf-b9a2-1244800b24da)

### Smartphone

```Markdown
- Device: Oneplus 9R
```

kripper, 2023-08-15:
Could you please confirm whether the video in QGC is always working fine without any quality issues, even after restarting multiple times?

I am experiencing quality issues with the video streaming to QGC from my Mini SE drone. If it consistently works well on QGC for you, we may be able to address the problem in gstreamer as well. However, in order to do so, we need to replicate your QGC environment first.
Could you let us know which version of QGC you are using?
Additionally, what video size are you utilizing in Rosetta Drone / DJI FlyApp?

Please provide as much information as possible about your setup.

jashshah999, 2023-08-16:
Hello. 

I am using QGC Version 4.2.8. Yes it is always working in QGC. I am 
always able to get a stream which has minimal lag, even after restarting multiple times.  This is using the stock version of rosetta, NO changes as per hacking on rosetta. Attaching below both the videos, in qgc and in gstreamer as well. 

In QGC, I am just inputting the input type as UDP H.264 and giving the port as 5600. That is all, and I am getting seamless streaming from QGC. 

However, when I use the command on gstreamer as follows, I do not get the same result. 

`gst-launch-1.0 -v udpsrc port=5600 ! application/x-rtp, encoding-name=H264 ! queue ! rtph264depay ! h264parse ! avdec_h264  ! autovideosink`

Please do let me know what is the possible reason for this. 

[Screencast from 08-15-2023 11:03:07 PM.webm](https://github.com/RosettaDrone/rosettadrone/assets/49280550/c47e7683-ec58-4672-85ff-ed1a7e27ba98)
[Screencast from 08-15-2023 11:05:14 PM.webm](https://github.com/RosettaDrone/rosettadrone/assets/49280550/bfbba8d2-a643-4043-a62e-4d1924466f24)

kripper, 2023-08-16:
Please confirm drone model, drone firmware and video resolution configured in Rosetta Stone / DJI Fly App.

kripper, 2023-08-16:
You can check and change the video resolution in DJI Fly App.
Maybe also in Rosetta Drone.
Any chance to test with latest QGC version on Windows?
I need to find out why it doesn't work on my setup with QGC.
I will try to reproduce.

jashshah999, 2023-08-17:
Let's talk on discord. How do I contact you?

kripper, 2024-04-10:
The best way to go is implement https://github.com/RosettaDrone/rosettadrone/issues/175

closed 2024-04-10 by kripper