RosettaDrone/rosettadrone #181: Can't connect UDP port from Rosettadrone App to QGC or Dronekit using DJI Mavic Air 2

jpa4094, 2023-09-08:
### Describe the bug

I have successfully managed to install the app from Android Studio. And it is able to successfully configure the DJI SDK product key. I even did the configurations for using external device for GCS and inserted the telemetry config. However, when i try to connect to the drone from dronekit or QGC, it doesn't seem to connect. Could there be something else i missed out? Kindly help me please. 

### Drone Model

```Markdown
DJI Mavic Air 2
```

### Smartphone Model

```Markdown
- Model:OPPO
- Android version:12
```

closed 2023-09-13 by jpa4094

kripper, 2023-09-13:
What was the problem?

jpa4094, 2023-09-13:
Once i used the same device to access rosettadrone, QGCS and Dronekit, the
UDP worked so no problem. Previously i used different devices that didn't
work.

However now i have a new issue. The drone is able to take off from dronekit
python script. But, no matter what altitute i set, the drone keeps hovering
at 1.2 metres only. I tried all functions that i can think of such as
channel override, rc_channel_raw, changing roll, pitch and throttle values
etc. However, the drone still stays there. Could you help me out in this? I
need to make the drone move left, right, forward, backward and upward.
Thanks in advance 🙏

kripper, 2023-09-13:
Use a debugger to see the logs.

jpa4094, 2023-09-13:
There is no error it seems. It ignores
 the commands given.

jpa4094, 2023-09-14:
Received unknown message. This is what i see in the logs as u can see in the screenshot[image]

jpa4094, 2023-09-14:
Alright will look through. By the way, just to ask if Rosettadrone has to be connected to QGC at the same time as well in order for the python script to work?

jpa4094, 2023-09-15:
I just saw the documentation and followed closely. It says received unknown message followed by MAV command. So do i need to edit the code on Android studio ?

kripper, 2023-09-15:
Yes. You should Google what messages are being sent by your python script, find out which ones you really need and implement them by looking how other messages are implemented.

You can also adapt your python script to use other messages 

Welcome aboard and have fun.

kripper, 2023-09-15:
> Alright will look through. By the way, just to ask if Rosettadrone has to be connected to QGC at the same time as well in order for the python script to work?

QGC is not need, but could be useful to monitor your drone.

jpa4094, 2023-09-15:
Thank you so much sir! You solved my issue that I have been troubleshooting for past 4 days. Yes I would implement vision algorithms as well and will definitely contribute as well!

kripper, 2023-09-15:
What did you do? Use other mavlink messages in your python script? In this case, which MAVLink msg was not working and which one worked?

jpa4094, 2023-09-18:
Sorry for late reply. Well, I tried with Mavlink commands like mavutil.mavlink.MAVLink_rc_channels_raw_message() didn't work. After trying 
def send_local_ned_velocity(vx, vy, vz):
    msg = vehicle.message_factory.set_position_target_local_ned_encode(
        0,
        0, 0,
        mavutil.mavlink.MAV_FRAME_BODY_OFFSET_NED,
        0b0000111111000111,
        0, 0, 0,
        vx, vy, vz,
        0, 0, 0,
        0, 0)
worked! I got this from one of the discussions that i read from this repository. 
Thanks.

jpa4094, 2023-09-19:
Btw, referring to the Hacking Rosetta document, to enable video stream from Rosetta to Dronekit, do we compulsory need linux version ? Does Windows don't support it ?

kripper, 2023-09-19:
The raw video streamer should work fine on any platform.

BTW, someone reported that RTP streaming was working fine with QGC for Linux, but not for Windows.
This should be investigated further.

jpa4094, 2023-09-19:
Alright i followed ediitng the PluginManager.java file and changing the hardcoded hostname to my PC ip address where i want the video to be streamed on. But when i go to the web browser and key in the ip address that i specified : 6000, the video still cannot stream.

kripper, 2023-09-19:
You didn't read the section "Receiving the Images directly in Python"

kripper, 2023-09-19:
And you hallucinated that your browser is capable of receiving this custom format that contains the drone angle :-)

Your next comment will probably be about the latency. Make sure to read other issues.

jpa4094, 2023-09-19:
Yes exactly you read my mind well :) I am reading your live editing on the document as well as reading other issues as well. Thanks

jpa4094, 2023-09-19:
I changed the number of frames from 15 to 6 to 3 and then to 1 currently since it has to be a divisor of 30. But i think now is slightly better with a delay of 2-3 seconds only.

jpa4094, 2023-09-20:
Btw Changing the code to improve video latency do u mean in Android Studio or Python script ?

kripper, 2023-09-20:
In both I guess.

jpa4094, 2023-09-24:
I have been trying to modify the Android Studio code to create a link between the Telemetry 3 host input and the host address hardcoded in the RawvideoStreamer file. I have tried intent method but it seems not working. Is there any method u would recommend me to use ?

kripper, 2023-09-24:
Telemetry ports and RawVideoStreamer ports are two completely different things and should not be linked.

jpa4094, 2023-09-28:
Alright, I have managed to create a new text input to add in and update the IP Host address and port of the raw video streamer from the app itself instead pf constantly updating the script and building it each and every time I use it at various locations.

jpa4094, 2023-09-28:
However, would like to check whether to run the vision landing and smart landing scripts, is it compulsory to use Linux and not Windows?

kripper, 2023-09-28:
I only developed and tested on Linux.
Please note the code to track the targets is written in C++.