RosettaDrone/rosettadrone #184: Drone keeps hovering on 1 metre for DJI Mavic Air 2 regardless of any other commands like simple_goto

jpa4094, 2023-09-14:
Dear Dronekit friends,

I've successfully connect Rosetta Drone and Dronekit, and use python script to interact with my Drone (Mavic Mini). It works with `simple_takeoff` command, but the altitude of my drone is stuck at 1 m although I set a higher altitude. Other command like `simple_goto` also not working, my drone doesn't move anywhere.

Any idea how to solve this issue? Thanks.

_Originally posted by @fajrulfx in https://github.com/RosettaDrone/rosettadrone/issues/36#issuecomment-878754085_

closed 2023-09-19 by jpa4094

kylenietfeld, 2024-08-31:
Same issue here. Is it because DJI requires at least two points for a waypoint mission? Anyone have a working example?

kripper, 2024-09-01:
Check the logcat.
It seems your drone is executing the take-off command and then stops executing other commands.

czarekk215, 2025-08-29:
@jpa4094 Did you manage to solve the problem?