RosettaDrone/rosettadrone #186: Mavic 2 Pro: No Takeoff or Video in QGC

kylenietfeld, 2023-10-09:
### Describe the bug

Hello, this may not be a bug but I am having trouble getting my Mavic 2 to do a takeoff or see video from the camera. 

Takeoff: I am able to succesfully connect to the Mavic 2 with RosettaDrone and it *connects* to QGC (the light is green and I can see some updating parameters such as battery life and # of satellites connected). QGC says 'Ready to Fly', however, when I try to takeoff, arm, or do an individual motor test, I get an error (one example attached). Is there something obvious that I am missing?  In my attempts, I did not have a higher number of GPS satellites connected.

Video: I tried the UDP options listed in QGC.

Any help would be greatly appreciated! 

### Screenshots

[image]

### Drone Model

```Markdown
Mavic 2 Pro
```

### Smartphone Model

```Markdown
- Model:TCL430M
- Android version: 13
```

kylenietfeld, 2023-10-09:
Also, the buttons on the left side of the screen of the RD app (besides the home button) are not operational. I'm guessing they should do something when clicked? 
[image]

kripper, 2023-10-10:
Have you checked the "Hacking on Rosetta Drone" guide?

kylenietfeld, 2023-10-14:
I have. I calibrated the compass using the DJI Go app and it said the drone is Ready to Go. Then, I connect with Rosetta and it auto connects to QGC. 

Rosetta should support the Mavic 2 Pro, correct? Is there anything else I need to do with my M2P before a takeoff command can be called? I noticed that QGC says there is 100% loss rate for the MAVlink commands, I'm sure that's part of the problem, but I'm not sure why. Is there any settings on QGC that need to be changed from the default settings?

kripper, 2023-10-15:
I would check the logcat using Android studio.
Maybe your firewall is blocking the UDP packets?

kylenietfeld, 2023-10-27:
Thanks for the suggestion! I noticed in the logcat in Android Studio that the line: 
sq.rogue.rosettadrone E/UDT-JNI: stopped udp

kept recoccuring during runtime. Looks like the problem but do you know what it means? I ran this with my firewall turned off.

stw2nf, 2023-10-30:
I am having this same problem with a Mini 2

kylenietfeld, 2023-11-04:
I noticed the log message when I send a command to my M2P is rc
mode == forbiddenModeSwitch == POSITION_ONE

The1only, 2023-11-05:
You have not unlocked Rosettadrone.Press the small lock up in the right corner.Best regardTerje Nilsen9Tek ASNorway

kripper, 2023-11-05:
> You have not unlocked Rosettadrone.Press the small lock up in the right corner

Maybe it's not that. I would suggest to check the code.
I remember there were some security restrictions (preventing some mode combinations) for specific models and comments explaining the reason.

Of course, something very simple to find and modify if you see the code.

People (including me) are a little scared to take a look at the code the first time. Then, they realize it's actually very easy to locate and change it.

kylenietfeld, 2023-11-05:
I am able to verify the video stream is being sent. I was able to see the video stream using gstreamer using the following command in the terminal: 

gst-launch-1.0 -v udpsrc port=5600 caps = "application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264, payload=(int)96" ! rtph264depay ! decodebin ! videoconvert ! autovideosink

I'm not sure why I am unable to see the video using QGC, but I don't need QGC for my application.

closed 2023-11-05 by kylenietfeld

kripper, 2023-11-06:
> rcmode == forbiddenModeSwitch ==

Search this string in the source code. It's the restriction I mentioned above.

kylenietfeld, 2023-11-06:
Thanks for your help in this! Can you explain a bit more about this (or point me to more information)? I saw that line in the code, but I didn't think there is any thing I can do about it. Am I able to remove it from the code? What purpose does this switch have?

kripper, 2023-11-06:
I'm not sure. Isn't there a comment?
Why didn't you copy/paste the code you found?

kylenietfeld, 2023-11-07:
when I search that string in this repository it returns these lines of code: 
<html>
<body>

private HardwareState.FlightModeSwitch rcMode = HardwareState.FlightModeSwitch.POSITION_ONE;
--
183 | private static HardwareState.FlightModeSwitch forbiddenModeSwitch = HardwareState.FlightModeSwitch.POSITION_ONE; //…
 
279 | if (m_aircraft == null \|\| !m_aircraft.isConnected()) {
 
286 | if (m_model == null) {
 
292 | forbiddenModeSwitch = HardwareState.FlightModeSwitch.POSITION_THREE;
 
296 | useMissionManager = m_model == Model.MAVIC_MINI
297 | \|\| m_model == Model.DJI_MINI_SE
298 | \|\| m_model == Model.DJI_MINI_2

</body>
</html>

Is this what you were referring to? Sorry, I am new to this.

kripper, 2023-11-07:
Yes. Somehow HardwareState.FlightModeSwitch.POSITION_THREE is forbidden under certain conditions.
I don't know why.
Maybe it's described in the commit or @The1only knows why.
Please use code formatting in the future.
The code block also looks incomplete.
Is useMissionManager used for this matter?

The1only, 2023-11-07:
The mode switches has different positions on Inspire 1, so pay attention to its setting.

kripper, 2023-11-07:
Actually, if I remember correctly, I had to disable this restriction for the mini versions, but I'm not sure. You will have to check the code.
Maybe we should just apply this restriction to certain models instead of apply it to all by default.

@The1only can you please explain more in detail why the mode needs to be restricted?