RosettaDrone/rosettadrone #189: DJI Mini 2 No Takeoff with QGC

stw2nf, 2023-11-01:
### Describe the bug

I have a Mini 2 connected to a Smart Controller running Rosetta Drone. I can successfully connect to the vehicle via UDP to QGroundControl, and open a vehicle connection Dronekit-python. A simple_takeoff command in Dronekit gets no response, and a takeoff command via QGC yields the error "Unable to takeoff. Vehicle failed to arm". Additionally any efforts to send velocity commands via Dronekit-python after manually arming and taking off don't produce any movement. The one command that does seem to work from both QGC and Dronekit is changing to Land mode.

Appreciate any advice! This is a great project and I am excited to utilize it.

### Screenshots

[image]

### Drone Model

```Markdown
DJI Mini 2
```

### Smartphone Model

```Markdown
- Model: DJI Smart Controller
- Android version: 7.1.2
```

stw2nf, 2023-11-01:
RTL seems to crash RosettaDrone app as well.

The1only, 2023-11-01:
And you armed Rosettadrone to allow for takeoff, and selected the  correct mavlink protocol i QGC?Best regardTerje Nilsen9Tek ASNorway

stw2nf, 2023-11-01:
@The1only I have unlocked the safety through Rosetta Drone. 

The mavlink protocol I am unsure about. Does this refer to when first booting QGC selecting the flight stack firmware you will be using, ArduPilot or PX4? In that case I believe I selected ArduPilot, but I will double check that is the case. Here are my Mavlink Settings through Application Settings on QGC though, and I just unchecked "Only accept MAVs with the same protocol version"

stw2nf, 2023-11-02:
@The1only So I did a reinstall of QGC and on that opening selection picked "Mavlink Generic. Have that "Only accept MAVs with the same protocol version" unchecked as well but still seeing the same behavior. 

Not sure if that was in line with what you were thinking the problem may be.

Thanks so much for the help! I really appreciate it.

stw2nf, 2023-11-03:
Similar to Issue #186 the buttons on the left side of RosettaDrone app are greyed out, I am receiving 100% loss rate in Mavlink messages and no video through QGC (Video streams fine through Mission Planner)

kylenietfeld, 2023-11-04:
I have same issue with Mavic 2 Pro. Not sure if I have some settings that are wrong. Readme and Hacking Rosetta Drone Guide don't have lots of information on QGC settings and makes me think it should work with default settings.

stw2nf, 2023-11-06:
@kylenietfeld Were there any changes that you made that allowed for takeoff? Are you building off master branch?

kylenietfeld, 2023-11-07:
@stw2nf I didn't do anything special. I just followed the instructions in the readme. I built the app, specified the IP address, connected to QGC, unlocked, and the takeoff/land worked. However, the same procedure is not working for my Mavic 2 Pro and I am not sure why

stw2nf, 2023-11-07:
@kylenietfeld In testing this I had a recent development. In C mode (Tripod in RosettaDrone) and S mode (Sport in RosettaDrone) if I give the vehicle a takeoff command, I get the same error "Vehicle failed to arm". However, if I then send a land command through QGC and then rapidly send another takeoff command as soon as possible through QGC the vehicle will arm and takeoff. I can't replicate this in N mode (P-GPS in RosettaDrone) though. I wonder if this sequence would work on your Mavic 2 as well? I am going to test on my Mini 2 from Dronekit-PYthon as well

kripper, 2023-11-08:
I remember take off from QGC used to fail the first time giving a "failed to arm" error. Setting to "auto" and retrying was the workaround I used.
I believe it was a problem with the checks performed by QGC.

In your case, you reported "100% loss rate in Mavlink messages".
Maybe a firewall blocking your UDP packets?
Make sure QGC is receiving messages sent by RosettaDrone.

BTW, You don't need to change anything related with MAVLink protocol version in QGC. I implemented V2 in RosettaDrone.

kripper, 2023-11-08:
I created: https://github.com/RosettaDrone/rosettadrone/issues/191

stw2nf, 2023-11-09:
@kripper Thanks a ton. In the flurry of debugging either I ended up doing what you recommended and arming in Auto mode accidently, or that Takeoff -> Land -> Takeoff sequence is another workaround for the Mini 2 /QGC combination. Regardless, after the first time, everything has been operating as expected. This is a fantastic project and hope to contribute where I am capable moving forward! Thanks for the help!

closed 2023-11-10 by stw2nf

juand5678, 2024-04-09:
@stw2nf  Before taking off, does the video stream work for you in the QGC?

kripper, 2024-04-10:
> @stw2nf Before taking off, does the video stream work for you in the QGC? I couldn't connect the video, could you help me? this is my email [diego.aristizabal@udea.edu.co](mailto:diego.aristizabal@udea.edu.co)

See https://github.com/RosettaDrone/rosettadrone/issues/175