RosettaDrone/rosettadrone #197: Very bad performance when trying to receive video from Mavic Mini

EMP-T, 2024-01-11:
### Describe the bug

I am trying to receive video from my Mavic Mini.
However, when using gStreamer (tried all pipelines from other issues), I am stuck with terrible image quality and mostly green blocks.
I tried using the RawVideoStreamer but I still end up with around with around 5-10 very distorted frames per second.
Even reducing the image to a resolution of 640x360 and greyscale has only increased the performance to about 15 FPS (but still with distorted content).
I have been running the receiver on both a Surface Book 2 (Ubuntu) and my desktop pc (Windows), both should have no issue receiving the stream.
In both the DJI Fly and RosettaDrone app the video looks perfectly fine.

### Screenshots

https://github.com/RosettaDrone/rosettadrone/assets/38190551/9130a923-9925-4502-8e50-278bb5738725

(Usually performance is better, but image quality is representative)

### Drone Model

```Markdown
DJI Mavic Mini
```

### Smartphone Model

```Markdown
- Model: OnePlus 6 & Samsung A53 5G
- Android version: 11 & 13
```

### Additional context

I would greatly appreciate any help as I would love to use RosettaDrone for a university project.

kripper, 2024-01-11:
> I have been running the receiver on both a Surface Book 2 (Ubuntu) and my desktop pc (Windows), both should have no issue receiving the stream.
> In both the DJI Fly and RosettaDrone app the video looks perfectly fine.

Not clear:
You write "Both should have no issue". Do they have issues?
DJY Fly and RD don't run on a windows desktop.

kripper, 2024-01-11:
Someone reported here recently that QGC had no issues with mavic mini on a Linux desktop, only on windows. It seems like something in the network layer is causing the issue.

EMP-T, 2024-01-11:
> Not clear: You write "Both should have no issue". Do they have issues? DJY Fly and RD don't run on a windows desktop.

I meant both laptop and PC should not be the bottleneck performance wise.

> Someone reported here recently that QGC had no issues with mavic mini on a Linux desktop, only on windows. It seems like something in the network layer is causing the issue.

I haven't even been able to get any video working in QGC unfortunately...

kripper, 2024-01-11:
> In both the DJI Fly and RosettaDrone app the video looks perfectly fine.

You mean the video works fine in DJI Fly and RosettaDrone, but not on your Laptop and PC?
That's a known issue.
All I can say is that one[ user reported that QGC was working fine on his Linux dekstop](https://github.com/RosettaDrone/rosettadrone/issues/174), which could be a hint to investigate a possible solution or workaround.

I believe the best approach to solve the problem is implementing https://github.com/RosettaDrone/rosettadrone/issues/175

closed 2024-01-11 by kripper