RosettaDrone/rosettadrone #198: SDK Registration failed

KroSmi, 2024-01-19:
### Describe the bug

After i generate the APK file from android studio 
[image]
and after installing the app on my android device i get this messages 
[image]
[image]
the package name in the dji developers sdk key is correct (sq.rogue.rosettadrone)
And can someone tell me which steps should i take to connect my android device to my dji air 2s when i fix this registration problem?

### Drone Model

```Markdown
DJI air 2s
```

### Smartphone Model

```Markdown
- Model:
- Android version: 12.0
```

KroSmi, 2024-02-04:
i still got the issue but i want to inform you that i have checked both API keys and they are both created and registered correctly

alexbhandari, 2024-03-24:
If your API key is correctly placed in the AndroidManifest file, package name is correct, and you have internet access on the phone it should connect (did for me). Did you also fill the Google maps API key?

For connecting to the drone, yes you just connect the controller to the phone with USB cable and the controller to the drone as usual.

KroSmi, 2024-03-25:
they are all correct but nothing is working :( . Can i get the apk from someone who has already has created the app ?

3iga, 2024-07-20:
I have a same issue. Did you find out a solution for it?

3iga, 2024-07-22:
> they are all correct but nothing is working :( . Can i get the apk from someone who has already has created the app ?
> 

I think, dji sdk is not supported in your region, because when i changed my region with vpn to france and created new account with new key, Rosetta started to work. However now i have problem with connection to qgs.

alexbhandari, 2024-08-19:
@3iga good point regarding the region - didn't think of that. Connection is where I am getting stuck as well. Not sure how to get it to connect and have put more details here: https://github.com/RosettaDrone/rosettadrone/issues/201