RosettaDrone/rosettadrone #199: Build error

juand5678, 2024-03-14:
My name is Juan and I am an aerospace engineering student from Colombia and I have been trying to install rosetta for 4 days but I have not been successful, could you help me with the installation? I already have the apy keys, and the program compiles but it does not execute, here I show you a screenshot of what I get

[image]

juand5678, 2024-03-14:
@kripper  podrias ayudarme con esto? mi correo es diego.aristizabal@udea.edu.co

kripper, 2024-03-15:
Parece que estás tratando de compilar para una arquitectura de hardware no soportada. Revisa tu target.

juand5678, 2024-03-15:
No e podido solucionar el error, podrías ayudarme?

-- 

"La información aquí contenida es para uso exclusivo de la persona o 
entidad de destino. Está estrictamente prohibida su utilización, copia, 
descarga, distribución, modificación y/o reproducción total o parcial, sin 
el permiso expreso de Universidad de Antioquia, pues su contenido puede ser 
de carácter confidencial y/o contener material privilegiado. Si usted 
recibió esta información por error, por favor contacte en forma inmediata a 
quien la envió y borre este material de su computador. Universidad de 
Antioquia no es responsable por la información contenida en esta 
comunicación, el directo responsable es quien la firma o el autor de la 
misma."

alexbhandari, 2024-03-24:
Did you install gitlfs per the instructions as the first step before cloning the repo? `Install git lfs as specified at https://github.com/git-lfs/git-lfs/wiki/Installation`

If you did not have this installed I suggest restarting and cloning the repo from scratch after you install it to get git to pull all the needed files

closed 2024-04-02 by juand5678