RosettaDrone/rosettadrone #209: MSDK V5 support

luzik, 2024-06-22:
### Is your feature request related to a problem? Please describe.

It would be awesome to have MSDK V5 support included with new gear supported like mini 3 pro

HEEJUN-SHIN, 2024-09-13:
Any updates on this??

LSC1759, 2024-12-15:
any hope?

angel-ayala, 2026-01-15:
Hi we are developing a new interface and aims to support the v5 SDK, the core idea is to support both versions of the SDK v4 and v5.
Initial test cases are focused on the v4 since is compatible with one aircraft we use at uni, however, the projected architecture should simplify a later migration to v5 somehow.
Everyone are welcome to join us in developing this application.

The project name is [WenuLink](https://github.com/WenuLink/wenu-link-android)