RosettaDrone/rosettadrone #214: How debug functionality for Phantom 4 Pro

angel-ayala, 2024-07-06:
Hi!,

I would like to use the Phantom 4 Pro (not v2.0 version) drone with RossetaDrone, I currently compiled and executed the apk in my mobile with no issues. When I plug the radio controller it recognizes both the controller and the product.
I already installed QGC on my laptop to work with RosettaDrone but I'm unable to make it fly.

On the APK side, I'm able to run the test mode and I receive the drone's camera and telemetry info. I changed the telemetry settings with my laptop's IP to connect to the QGC. In test mode the QGC recognizes the connection but I does not got any video image.
When I use the simulation mode also recognizes the connection, but no other interaction can be done.
When I use the Drone mode, the QGC recognizes the connection but I'm not able to take off or receive video.

On the APK's log for the simulation and Drone mode I got the "Do NOT Transcode Video !!!!!!!" message.

I already checked the logcat from the android studio but there is a lot of information and I don't know how filter it or which kind of message search for, also always show the "Empty SMPTE 2094-40 data" error.

Can you give me any orientation of how can I debug the or config any piece of code? Thanks!

closed 2024-07-21 by angel-ayala

kripper, 2024-07-21:
Nice. For the video, see https://github.com/RosettaDrone/rosettadrone/issues/175

angel-ayala, 2024-07-21:
I updated the issue with the template in a new one. Can you take a look? Should be the same issue for video?