RosettaDrone/rosettadrone #215: APK Build

timotoots, 2024-07-15:
### Is your feature request related to a problem? Please describe.

It is very difficult for non-android programmer to set everything up so the project builds correctly.

### Describe the solution you'd like

Could it be possible to change the project so it could be built without developer keys and later add them in config?

### Describe alternatives you've considered

Or maybe create workflow for github, so it's easier to build.

timotoots, 2024-07-16:
I have tried to create a workflow that would end up with a APK package. It seems mostly to work. I forked the repo and tuned the workflow file, it does result in APK.

But now the problem is substituting the keys. It's not good to store keys in a public repo, so I need to do it private. But it is not possible to change it as it's a forked repo.

The other solution could be using Github Actions secrets functionality, but it does also not work with forked repos. 

Any ideas how to solve this part? If we get it working, we could update README and add the workflow to this repository for others for easier building.

kripper, 2024-07-16:
This project is focused on developers.
If you release an APK with your keys, please also give the users support and close their issues.

closed 2024-07-16 by kripper

timotoots, 2024-07-16:
I would like to run it myself and I believe there are other that would not develop it, but test it in the real world. If it's made easier that navigating Android dependencies, it would make this more usable.
Is there somebody who has released this as an ready-made app?

kripper, 2024-07-16:
We are not interested in testers or users that are not able/willing to develope and contribute with pull requests.
We don't want to have non-developers reporting issues or asking for support in our issues section.

kripper, 2024-07-16:
There is no roadmap nor plans to support end users.
It's a repo where developers share their work with other developeres.

timotoots, 2024-07-16:
Thanks for answering.

Just final thoughts. The features that this repo provides, could be very interesting for developers who could not develop Android apps, but use it to develop other software packages. So at the moment it seems to be already usable for that, just that not so easy if not familiar with Android ecosystem.

I guess I just have to push through, get it compiled to be able to use it for something interesting.