RosettaDrone/rosettadrone #218: Phantom 4 Pro problem with MAVLINK commands

angel-ayala, 2024-07-22:
### Describe the bug

After source code compilation and installing RD, the APK crashed when after tap the Drone button.
I changed two lines of code, and I was able to use it, however, I still facing some problems.

I already set the RD and QGC settings, and I can arm the drone from my smartphone and through QGC, nevertheless, no success with other MAVLINK commands.
With QGC I can read some sensor data, but I can't send the home location or other MAVLINK related commands with success.

On the logcat from android studio, after open RD I get some some error log, then when I tap the Drone button I get other error messages.

When I start the [DroneKit's python script](https://github.com/dronekit/dronekit-python/blob/master/examples/vehicle_state/vehicle_state.py) to read the vehicle state, I can see the captured state until the armed info, after that, the script keeps waiting set the home location command, which never happens.

There are some LDM related error, but on the DJI's site appears that is not mandatory, maybe because is a chinese smartphone?
I don't know, can you help me?

### Steps To Reproduce

1. Install following the installation instructions.
2. Add `case PHANTOM_4_PRO:` in the `MainActivity.java` file at line 949, and update the line 296 of `DroneModel.java` with `useMissionManager = m_model == Model.PHANTOM_4_PRO`.
3. Connect the RC by USB to the smartphone with the drone connected.
4. Start the apk from AndroidStudio.
5. Tap Drone button.
6. Start DroneKit [vehicle_state.py](https://github.com/dronekit/dronekit-python/blob/master/examples/vehicle_state/vehicle_state.py) example script.

### Drone Model

```Markdown
DJI Phantom 4 Pro
```

### Smartphone Model

```Markdown
- Model: Xiaomi Redmi Note 10S (M2101K7BL)
- Android version: 13 TP1A.220624.014
```

### Additional context

DroneKit `python vehicle_state.py --connect udpin:0.0.0.0:14550` command output:
```
Connecting to vehicle on: udpin:0.0.0.0:14550

Get all vehicle attribute values:
 Autopilot Firmware version: APM:Copter-0.4.1-dev7
   Major version number: 0
   Minor version number: 4
   Patch version number: 1
   Release type: dev
   Release version: 7
   Stable release?: False
 Autopilot capabilities
   Supports MISSION_FLOAT message type: False
   Supports PARAM_FLOAT message type: False
   Supports MISSION_INT message type: True
   Supports COMMAND_INT message type: True
   Supports PARAM_UNION message type: False
   Supports ftp for file transfers: False
   Supports commanding attitude offboard: False
   Supports commanding position and velocity targets in local NED frame: False
   Supports set position + velocity targets in global scaled integers: False
   Supports terrain protocol / data handling: False
   Supports direct actuator control: False
   Supports the flight termination command: False
   Supports mission_float message type: False
   Supports onboard compass calibration: False
 Global Location: LocationGlobal:lat=0.0,lon=0.0,alt=None
 Global Location (relative altitude): LocationGlobalRelative:lat=0.0,lon=0.0,alt=0.0
 Local Location: LocationLocal:north=None,east=None,down=None
 Attitude: Attitude:pitch=-0.003490658476948738,yaw=-0.6824237108230591,roll=-0.006981316953897476
[... 215 lines cut ...]
2024-07-21 19:21:56.488 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.488 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglSwapBuffersWithDamageKHRImpl:1372 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.531 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.534 17549-17765 Surface                 sq.rogue.rosettadrone                E  freeAllBuffers: 1 buffers were freed while being dequeued!
2024-07-21 19:21:56.534 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglSwapBuffersWithDamageKHRImpl:1372 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.538 17549-18166 Surface                 sq.rogue.rosettadrone                E  getSlotFromBufferLocked: unknown buffer: 0xb400007c0c73e100
2024-07-21 19:21:56.564 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.630 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.634 17549-17765 Surface                 sq.rogue.rosettadrone                E  freeAllBuffers: 1 buffers were freed while being dequeued!
2024-07-21 19:21:56.635 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglSwapBuffersWithDamageKHRImpl:1372 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.637 17549-18166 Surface                 sq.rogue.rosettadrone                E  getSlotFromBufferLocked: unknown buffer: 0xb400007c0c73e700
2024-07-21 19:21:56.663 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.729 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.760 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.765 17549-17765 Surface                 sq.rogue.rosettadrone                E  freeAllBuffers: 1 buffers were freed while being dequeued!
2024-07-21 19:21:56.765 17549-18166 libEGL                  sq.rogue.rosettadrone                E  eglSwapBuffersWithDamageKHRImpl:1372 error 300d (EGL_BAD_SURFACE)
2024-07-21 19:21:56.767 17549-18166 Surface                 sq.rogue.rosettadrone                E  getSlotFromBufferLocked: unknown buffer: 0xb400007c0c89ec00
2024-07-21 19:21:56.825 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.829 17549-17765 Surface                 sq.rogue.rosettadrone                E  freeAllBuffers: 1 buffers were freed while being dequeued!
2024-07-21 19:21:56.831 17549-18166 Surface                 sq.rogue.rosettadrone                E  getSlotFromBufferLocked: unknown buffer: 0xb400007c0c89ed80
2024-07-21 19:21:56.895 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
2024-07-21 19:21:56.898 17549-17765 Surface                 sq.rogue.rosettadrone                E  freeAllBuffers: 1 buffers were freed while being dequeued!
2024-07-21 19:21:56.901 17549-18166 Surface                 sq.rogue.rosettadrone                E  getSlotFromBufferLocked: unknown buffer: 0xb400007c2a7ecc80
2024-07-21 19:21:56.929 17549-17765 libEGL                  sq.rogue.rosettadrone                E  eglMakeCurrentImpl:1082 error 3002 (EGL_BAD_ACCESS)
```

angel-ayala, 2024-07-29:
Apparently MAVLink is working fine, nevertheless there are some `MAV_CMD_REQUEST_MESSAGE` that may need proper handlers.

Every time when I have the QGC and start RD, the app crash with the following exception.

The issue should be with the FTP transaction, according to MAVLink documention, requestMsgId 395 is `[COMPONENT_INFORMATION](https://mavlink.io/en/messages/common.html#COMPONENT_INFORMATION)` which is a 

> Component information message, which may be requested using [MAV_CMD_REQUEST_MESSAGE](https://mavlink.io/en/messages/common.html#MAV_CMD_REQUEST_MESSAGE).

Apparently that FTP transaction done is being done with a null pointer, additionally the method appears as deprecated since 2022-04.

Any ideas how to solve it?

```
2024-07-29 15:43:05.568 10209-10637 MAVLinkReceiver         sq.rogue.rosettadrone                D  MAV_CMD_REQUEST_MESSAGE -> requestMsgId=148
2024-07-29 15:43:06.614 10209-10637 MAVLinkReceiver         sq.rogue.rosettadrone                D  MAV_CMD_REQUEST_MESSAGE -> requestMsgId=397
2024-07-29 15:43:07.637 10209-10637 MAVLinkReceiver         sq.rogue.rosettadrone                D  MAV_CMD_REQUEST_MESSAGE -> requestMsgId=395
2024-07-29 15:43:08.663 10209-10637 AndroidRuntime          sq.rogue.rosettadrone                E  FATAL EXCEPTION: Thread-191
                                                                                                    Process: sq.rogue.rosettadrone, PID: 10209
                                                                                                    java.lang.NullPointerException: Attempt to invoke virtual method 'void sq.rogue.rosettadrone.FTPManager.manage_ftp(com.MAVLink.common.msg_file_transfer_protocol)' on a null object reference
                                                                                                    	at sq.rogue.rosettadrone.MAVLinkReceiver.process(MAVLinkReceiver.java:521)
                                                                                                    	at sq.rogue.rosettadrone.MainActivity$GCSCommunicatorAsyncTask$Listener.run(MainActivity.java:1773)
```

kripper, 2024-07-29:
Definetively a bug in the implementation.

kripper, 2024-07-30:
Why is it using FTP? Maybe it's old custom code that shouldn't be there.
One of the previous authors of RD (@The1only) was doing some custom FTP operations, like uploading stats to a private FTP server.
His customizations should be removed from the main code and placed into the "AI9Tek" plugin `(app/src/main/java/sq/rogue/rosettadrone/plugins/AI9Tek.java),` as I have done myself in the past.
The mentioned plugin is broken. I moved the code there only to make him easier to re-enable his customizations in case he needs them in the future.

The bug you described could be triggered by some leftovers of his customizations in MAVLINK command that were not tested afterwards.
If so, feel free to remove those customizations and insert them as comments in the "AI9Tek" plugin.

kripper, 2024-07-30:
I also found and fixed many of those missing ACKs. I don't remember if QGC and MAVSDK were 100% compatible regarding ACKs (I remember the implementations had some deviations, but I got them working fine for my use cases).

angel-ayala, 2024-09-04:
I already did some tests, and when I create a mission on QGC appears the uploading message in logcat but when slide the bar to start the mission, nothing happens after that.
I will look more in depth to what is wrong, at first, should be something related to MAVLink only.

kripper, 2024-09-04:
> I already did some tests, and when I create a mission on QGC appears the uploading message in logcat but when slide the bar to start the mission, nothing happens after that. I will look more in depth to what is wrong, at first, should be something related to MAVLink only.

What DJI model?

kripper, 2024-09-04:
I'm not sure if phantom 4 supports on-board missions.
I implemented and tested extensively the `MissionManager` class to support off-board missions (handled by RD).
If your drone doesn't support on-board missions, you can enable the `MissionManager` class by adding the model in the list of models (one line of code).
For more info, please search the issues and code by `MissionManager`.

angel-ayala, 2024-09-04:
I don't exactly know if P4P supports on-board missions, but I was looking into the `MAVLinkReceiver` class, there are some if related with the `MissionManager` and check if current model supports on-board mission.

angel-ayala, 2024-09-06:
There is anyway to test the support of on-board missions? I accidentally already added the model to the list and enabled it.
When I upload the mission enters into `MAVLinkReceiver` class's `finalizeNewMission` method with all working fine, but after sliding the drone lifts and then nothing happens and, there is no movement nor error log.

After populating the `missionItems` through `mModel.missionManager.setMission` which method (must be) is called? I was thinking in check the method's flow calls first, I'm using the mission planner from QGC with just one coordinate.
Any thoughts on what could it be?

kripper, 2024-09-06:
For onBoard, just don't use the MissionManager class.
I couldn't test it though because I have a cheap mini.