RosettaDrone/rosettadrone #220: Camera feed is compatible with MAVLink?

angel-ayala, 2024-08-14:
### Is your feature request related to a problem? Please describe.

Hi, I was wondering if the camera feed are in the MAVLink protocol, to have a centralized sensor communication as standard.
I found this [repo](https://github.com/mavlink/mavlink-camera-manager) and it seems interesting to be appearing the drones video feed.

### Describe the solution you'd like

- Check the [camera service](https://mavlink.io/en/services/camera.html) implementation in current code.
- Check the compatibility with the WebRTC protocol.

### Additional context

As mentioned in #219, that the video feed freezes when the YUV data is decoded and transmitted. Maybe can be assured a full MAVLink compatibility, proxying the drone sensors and actuators through the protocol.

kripper, 2024-08-14:
I don't know if the MAVLink messages regarding video streaming are mature or supported by QGC.
I would test gstreamer with WebRTC and provide the QGC team with the gstreamer command-line to support RD + WebRTC.
And also ask them if they support some MAVLink messages to setup QGC's video stream receiver.

kripper, 2024-08-14:
I'm not sure If I understood correctly your issue.

angel-ayala, 2024-08-16:
> I don't know if the MAVLink messages regarding video streaming are mature or supported by QGC.

I also don't know that, I just was wondering if the current code of RD has enabled the [camera service](https://mavlink.io/en/services/camera.html) to centralize the sensor readings. My idea is to get all readings through a python script using DroneKit-Python. If QGC supports it is also an open question.

However, I was thinking that maybe the [MAVLink camera manager](https://github.com/mavlink/mavlink-camera-manager) could be used to check the camera services availability.

kripper, 2024-08-16:
I implemented some camera trigger and movement mavlink messages for the MissionManager class, but I didn't do anything with the camera discovery messages.

If you need them, it's good to implement the corresponding mavlink messages.

If the mavlink messages definition is incomplete, you can contact the mavlink team and propose adding new messages and parameters.