RosettaDrone/rosettadrone #221: Dependency problem WebRTC and mapbox

utn-blei, 2024-09-04:
### Describe the bug

Hello, 
thanks for the great project!
Buildin the app unfortunately results in the bug, shown above.
Any idea how to fix it?
Greeting!

### Steps To Reproduce

1. Clone Repository
2. Adjust jdk versions
3. Hit "Build Project"

### Screenshots

[image]

### Drone Model

```Markdown
DJI Air 3
```

### Smartphone Model

```Markdown
- Model: emulator
- Android version:
```

kripper, 2024-09-04:
@angel-ayala can you please take a look?

angel-ayala, 2024-09-04:
I actually didn't test it in a emulator, but I don't know how the UI could be related. Additionally, the error says that no package was found in any repo. I have the exact gradle config from [master](https://github.com/RosettaDrone/rosettadrone/blob/master/app/build.gradle) if no diff could be a connection issue, maybe?

> Adjust jdk versions

Just to be sure @utn-blei, the adjustement were those in #172 right? which OS are you using?

kripper, 2024-09-04:
RD only works on specific architectures, because it depends on the DJI SDK.

closed 2024-09-04 by kripper

utn-blei, 2024-09-04:
Thanks for the quick reply!

Here is my workflow: 
- git clone the project
- import the project to android studio
- change the versions like in #172  
- waiting for the sync to happen
-> I already obtain warnings about missing dependencies (see screenshot)

- I then try to build -> this leeds to the error seen above

To the best of my knowledge, the probelm should not be related to the emulator. Right?

[image]

angel-ayala, 2024-09-04:
there are a [few post](https://stackoverflow.com/questions/47435348/dji-mobile-sdk-android-studio-emulator-sdk-registration-failed) indicating the emulator thing

utn-blei, 2024-09-04:
Thanks :) I cant even generate any .apk though since it fails much earlier in the process.
I can therefore also not test on my mobile device.

How would you recommend building the project?

utn-blei, 2024-09-04:
Seems like others face the same issue - at least regarding mapbox:
https://github.com/RosettaDrone/rosettadrone/issues/172#issuecomment-2295017779

HEEJUN-SHIN, 2024-09-07:
hello, @utn-blei. I faced the same problem regarding the mapbox and webrtc-google.

Does the https://stackoverflow.com/questions/68934036/could-not-find-com-mapbox-mapboxsdkmapbox-android-accounts0-7-0-on-fresh-inst help the mapbox problem? Should I put the API key inside the password?

I found out the problem comes from 'com.dji:dji-uxsdk:4.16', so I temporally added 
exclude group: 'com.mapbox.mapboxsdk', module: 'mapbox-android-accounts'
below the implementation and It could be built successfully.

HEEJUN-SHIN, 2024-09-07:
Also, the "Could not find org.webrtc:google-webrtc:1.0.32006" error, it seems like there are no actual file in the maven repository. 

so I tried to look for webrtc-sdk in google repository, and found 
https://github.com/webrtc-sdk/android

using implementation 'io.github.webrtc-sdk:android:125.6422.05'
instead of implementation 'org.webrtc:google-webrtc:1.0.32006' seems resolving the issue.

But I think there should be more changes on the settings.

yblei, 2024-09-07:
Yes, adding the mapbox token resolved the issues. 
I followed the official guid on webboxes wesite though. 

Regarding webrtc: I picked an earlier commit before it was integrated to build the app.

yblei, 2024-09-07:
> Also, the "Could not find org.webrtc:google-webrtc:1.0.32006" error, it seems like there are no actual file in the maven repository.
>
> so I tried to look for webrtc-sdk in google repository, and [...]

Great!! Thanks for sharing.

kripper, 2024-09-07:
Please provide a pull request. Thanks.