RosettaDrone/rosettadrone #222: Outdated WebRTC Version breaks build

mjohenneken, 2024-09-10:
### Describe the bug

Hey folks,
checking out this repository i ran into some issues with the build scripts. Running with the Android Studio 2020 version the app can be built with the outdated gradle and android build tools versions. However the google-webrtc dependency introduced the lastest  PR is not outdated and not maintained. Additionally Jcenter shutdown and redirects the requests to Maven Central (see https://jfrog.com/blog/jcenter-sunset/) which doesn't host the package.

```
Could not find org.webrtc:google-webrtc:1.0.32006.
Required by:
    project: app
```

I'd suggest to use a different webrtc dependency. e.g. https://github.com/GetStream/webrtc-android?tab=readme-ov-file or the one from Twilio

### Drone Model

```Markdown
N/A
```

### Smartphone Model

```Markdown
- Model:
- Android version:
```

kripper, 2024-09-10:
Thanks.
@angel-ayala can you take a look please?

mjohenneken, 2024-09-10:
Quoting From the[ JCenter sunset note](https://jfrog.com/blog/jcenter-sunset/):
> **What should a user do if a package they need is not available in Maven Central?**
> If a specific package version you need is not available on Maven Central, you have the following options:
> 
> - Contact the package owner and request that they make it available on Maven Central
> - Check for a more recent version of the package on Maven Central
> - Remove your dependency on the package
> - Copy the package to your organization’s internal repository
> If you are a JFrog user, and none of these options work for you, please reach out to support@jfrog.com.

For webrtc its seems we can use this package: 
```
dependencies {
    implementation("io.getstream:stream-webrtc-android:1.1.3")
}
```
But also i found that the mapbox-android-sdk:9.2.0 links to jcenter packages which are not available anymore. I guess you still have a local copy in the package cache if you had a successful build before Aug 15th
```
Could not find com.mapbox.mapboxsdk:mapbox-android-accounts:0.7.0.
Required by:
    project :app > com.mapbox.mapboxsdk:mapbox-android-sdk:9.2.0
```

mjohenneken, 2024-09-10:
https://github.com/mapbox/mapbox-gl-native-android/issues/721 Mapbox sdk is archived and there seem to be no plans to release those sdks on mavencentral. I guess the dependency need to be updated  to make it work again for new users of rosettadrone
Found the related issue created by @angel-ayala at the DJI MSDK https://github.com/dji-sdk/Mobile-UXSDK-Android/issues/154

kripper, 2024-09-10:
There is another explanation, but I don't want to mix different issues here.

kripper, 2024-09-10:
This issue was for @angel-ayala (the webrtc issue), but since you are mixing things here and the title is not clear anymore, I'm closing this one.

closed 2024-09-10 by kripper

kripper, 2024-09-10:
Are you sure you provided a Google Maps API key as explained in the README?

angel-ayala, 2025-08-30:
Sorry for the late reply, by the moment I was taking some day off and totally forgot to reply.
I was looking tom implement WebRTC in another Android project, and I found out something related to a deprecated Android version of the library.
Check [this link](https://stackoverflow.com/questions/72103146/how-to-fix-failed-to-resolve-org-webrtcgoogle-webrtc) and [this other](https://groups.google.com/g/discuss-webrtc/c/Ozvbd0p7Q1Y/m/M4WN2cRKCwAJ) one.

As solution can be used the jcenter or add it as an AAR file inside the project, thats why maybe exists [this repo](https://github.com/rno/WebRTC) too, don't know, just a though.
Additionally, it can be compiled to create the corresponding AAR file you need for the architecture, as mentioed [here](https://webrtc.googlesource.com/src/+/refs/heads/main/docs/native-code/android/).

Maybe related to #221