RosettaDrone/rosettadrone #223: recently added webrtc stuff breaks build

tednv, 2024-09-19:
### Describe the bug

```
rosettadrone\app\src\main\java\sq\rogue\rosettadrone\plugins\WebRTCStreaming.java:60: error: m_model is not public in DroneModel; cannot be accessed from outside package
            if (pluginManager.mainActivity.mModel.m_model == null) {
                                                 ^
rosettadrone\app\src\main\java\sq\rogue\rosettadrone\plugins\WebRTCStreaming.java:68: error: m_model is not public in DroneModel; cannot be accessed from outside package
                        pluginManager.mainActivity.mModel.m_model, stunServer);
                                                         ^
```

### Steps To Reproduce

Checkout master, build

### Drone Model

```Markdown
mavic 2 pro
```

### Smartphone Model

```Markdown
- Model:
- Android version:
```

### Additional context

tried several different gradle and java versions to see if it's a compatibility issue, all fail consistently with the same error

kripper, 2024-09-19:
Just make it public?

@angel-ayala maybe some required changes were not included in the pull request?

tednv, 2024-09-19:
tried making public, which does get apk release to build but then when running on phone it crashes as soon as video feed begins so there are probably other things that were missing from the pr

kripper, 2024-09-19:
Take a look at the android logcat logs.
See: https://github.com/RosettaDrone/rosettadrone?tab=readme-ov-file

angel-ayala, 2024-09-19:
Oh yes, I forgot to update `DroneModel` class, as I did some other changes to use with the P4P I did not push those changes, I will make the  changes in my repo and then the PR

closed 2024-09-19 by kripper

tednv, 2024-09-20:
awesome, works now, thank you for quick fix!

PR #224 (2024-09-19): making public  m_model in DroneModel.java
Update in `DroneModel` class to fix #223