RosettaDrone/rosettadrone #225: Cannot send set position command in NED

adrian-carrio, 2024-10-07:
### Describe the bug

Hello, 

I am trying to fly a Mavic Air 2S in an indoor environment using local coordinates. I am able to take off and land autonomously without a problem using mavlink commands with python scripts. However, when I send a command like the one below the drone does not move at all. I have tried with various bit masks without success. 

set_position_target_local_ned_encode(
        0,
        0, 0,
        mavutil.mavlink.MAV_FRAME_LOCAL_NED,
        bit_mask,
        x, y, z,
        0, 0, 0,
        0, 0, 0,
        0, 0)
        
Any help is very much appreciated.        

### Drone Model

```Markdown
Mavic Air 2S
```

adrian-carrio, 2024-10-24:
It is indeed implemented, simply not working. Has anyone been able to send this command?
Also, using the virtual sticks demo app from DJI's SDK I can access the local pose but I cannot access the local pose using mavlink. Has anyone been successful here? Thank you

kripper, 2024-10-24:
It's probably not correctly implemented.
Pull requests are welcome.

kripper, 2024-10-24:
Should be quite easy to fix though.

BTW, I implemented it [here](https://github.com/RosettaDrone/rosettadrone/blob/46175575233f8556d263f3b74447503b7533d9ca/app/src/main/java/sq/rogue/rosettadrone/MAVLinkReceiver.java#L344) and it probably was working for my use cases.

Use RD in test mode and debug using Android Studio.

adrian-carrio, 2024-10-25:
Thank you. Do you know whether it could work without GPS signal? Is there some way to know this other than trial & error?

kripper, 2024-10-25:
We use the SDK to get the "GPS location", but the firmware is probably returning a EKF filtered location merging GPS + accelerometer + CV and other inputs. 

Depending on your model and firmware, your aircraft may even reject to take off is the signal is too weak.

RD is also checking the signal and rejecting any action if the GPS signal is too weak.

GPS is normally weak or takes some time to calibrate when not open air.

The GPS signal is displayed on RD and there are messages on the log (logcat).

In theory, you could modify RD and implement a 
ORB SLAM algorithm to navigate only using the camera input without GPS if that's what you want.