RosettaDrone/rosettadrone #228: Yaw turn MAV_CMD_CONDITION_YAW

luisgrana2, 2025-01-09:
### Describe the bug

I am working with Rosetta for autonomous flies with MAVIC AIR 2S. The goal is to perform a 180-degree yaw turn after takeoff. I have seen that the CONDITION_YAW command is implemented, but it does not seem to execute as expected when flying indoors without a GPS signal.

Takeoff and landing are functioning correctly, but the drone does not perform the yaw turn.

from pymavlink import mavutil
import time

connection_string = 'udp:xxx.xxx.x.xx:xxxxx'
print(f"Connecting to vehicle on: {connection_string}")

try:
    # Create the MAVLink connection object
    master = mavutil.mavlink_connection(connection_string)

    # Wait for a heartbeat to confirm the connection
    print("Waiting for heartbeat from the vehicle...")
    master.wait_heartbeat()
    print(f"Heartbeat received from system {master.target_system}, component {master.target_component}")

    # Function to set the flight mode of the drone
    def set_mode(mode):
        """Change the flight mode of the drone."""
        mode_id = master.mode_mapping()[mode]
        master.mav.set_mode_send(
            master.target_system,
            mavutil.mavlink.MAV_MODE_FLAG_CUSTOM_MODE_ENABLED,
            mode_id
        )
        
        # Confirm the mode change
        while True:
            msg = master.recv_match(type='HEARTBEAT', blocking=True)
            if msg.custom_mode == mode_id:
                print(f"Mode successfully set to {mode}.")
                break

    # Function to command the drone to take off
    def takeoff(target_altitude):
        """
        Commands the drone to take off to a specified altitude.
        :param target_altitude: Target altitude in meters.
        :param writer: CSV writer object to log metadata.
        """
        print(f"Taking off to {target_altitude:.2f} meters...")

        # photo_count = 0

        # Send takeoff command
        master.mav.command_long_send(
            master.target_system,
            master.target_component,
            mavutil.mavlink.MAV_CMD_NAV_TAKEOFF,
            0,
            0, 0, 0, 0, 0, 0,  # Unused parameters
            0
        )

        time.sleep(1)

        while True:
            # Trigger the camera and log metadata

            # Request altitude data
            master.mav.request_data_stream_send(
                master.target_system,
                master.target_component,
                mavutil.mavlink.MAV_DATA_STREAM_POSITION,
                1,  # Frequency (Hz)
                1   # Enable stream
            )

            # Wait for altitude data
            msg = master.recv_match(type='GLOBAL_POSITION_INT', blocking=True)
            if not msg:
                print("Not height message")
                continue
            

            current_alt = msg.relative_alt / 1000.0  # Convert millimeters to meters
            print(f"Current Altitude: {current_alt:.2f} m")

            # Log metadata

            # Stop when near target altitude
            if current_alt >= (target_altitude * 0.5):
                print("Target altitude reached.")
                break

            time.sleep(.5)

    def yaw_turn(target_heading = 180, direction = 1, is_relative = 1):

        # Send orientation command
        master.mav.command_long_send(
            master.target_system,  # system ID
            master.target_component,  # component ID
            mavutil.mavlink.MAV_CMD_CONDITION_YAW,  
            0,  # Confirmation
            target_heading,  # param1: Degrees
            0, #yaw_speed,       # param2: Velocity
            direction,       # param3: Direction
            is_relative,     # param4: Absolut or relative
            0, 0, 0          # param5-param7: Not used
        )

        time.sleep(5)

    if __name__ == "__main__":

        set_mode('GUIDED')  # Set mode to GUIDED before arming

        print("Starting photo capture with altitude metadata...")
        takeoff(1.5)  # Execute takeoff
        print("Video init complete.")

        # goto_altitude(3.0)

        yaw_turn(180)

        # Command the drone to land
        set_mode('LAND')
        print("Drone landing...")

except KeyboardInterrupt:
    print("Mission interrupted by user.")
except Exception as e:
    print(f"An error occurred: {e}")
finally:
    print("Closing MAVLink connection.")
    if 'master' in locals():
        master.close()
        
        
Any idea why the drone is not turning in yaw? I'm new to developing this type of application. Sorry :)

### Drone Model

```Markdown
DJI MAVIC AIR 2S
```

kripper, 2025-01-09:
Please update and re-test. I just commited a bug fix.
If it still doesn't work, please check the documentation to debug RosettaDrone.

kripper, 2025-01-09:
I also highly recommend to first test your code with the "Test" mode (DummyProduct class).