RosettaDrone/rosettadrone #229: "Failed to arm vehicle" on Takeoff from QGC

antonibote, 2025-01-27:
Referencing issue https://github.com/RosettaDrone/rosettadrone/issues/189 I am also using a setup with a DJI Mini 2 and QGC, where the drone fails to arm and consequently takeoff (100% Mavlink loss rate as well) . None of the workarounds mentioned in the issue seem to work for me. Since it is a little bit old, I was wondering if a solution has been found to this issue. 

Thank you

kripper, 2025-01-27:
Please don't duplicate issues.

kripper, 2025-01-27:
- Check if there are warnings displayed by DJI Fly app
- Retry twice
- Disable safety mode
- Use Android studio to examine logs and trace the execution to find out what is failing. See https://docs.google.com/document/d/1wR56Mvd2OsT82kxMOD7f_3-c6yJGBtLaT8LuVrmC5wI/edit?usp=sharing

antonibote, 2025-01-27:
Thank you for the quick response, it is much appreciated. 

I have a DJI Mini 2 connected to its controller which is running RosettaDrone. I can control the drone directly via the controller. Additionally, I have set up a telemetry port to connect to QGC. The connection is established correctly but it all comes down to trying to take off via dronekit script or QGC, where I get the error: "Failed to arm vehicle". Moreover, I get a 100% loss rate on Mavlink commands in QGC.  From the issue, I have checked: 

- Unlock safe mode
- Setting to auto and retrying
- Takeoff -> Land -> Takeoff sequence

kripper, 2025-01-27:
Did you check your firewall on the QGC box?
Use Android studio to check what's going on.

kripper, 2025-01-29:
I remembered something very important: I noticed that every time I tried to take off via the SDK, my Mavic Mini refused at the beginning.
But after I took off with the joystick first, it kinda unblocked this undocumented protection or whatever it is.

Other times, I had to switch to DJI Fly to see the alerts preventing to take off.

But in your case, you probably have a networking problem.